aqualinkd/AqualinkD #285: V2.3.4 make fails on bullseye -  multiple definition of `_commands'

ballle98, 2024-05-02:
Similar to issue #185.  With older versions of gcc you could have duplicate definitions of global variables as long as they are only reference in the files they are defined in.  This is no longer allowed and you need to make them static

```
pi@raspberrypi:~/git/AqualinkD $ make
OS: 11 (bullseye)
GLIBC build with: ldd (Debian GLIBC 2.31-13+rpt2+rpi1+deb11u8) 2.31
GLIBC Prefered  : 2.24-11+deb9u1 2.24
gcc -Wall -O3  -D AQ_PDA -D AQ_ONETOUCH -D AQ_IAQTOUCH -D AQ_RS16 -D AQ_MANAGER -D MG_ENABLE_HTTP_SSI=0 -D MG_ENABLE_DIRECTORY_LISTING=0 -D MG_ENABLE_HTTP_CGI=0  -o release/aqualinkd build/aqualinkd.o build/utils.o build/config.o build/aq_serial.o build/aq_panel.o build/aq_programmer.o build/net_services.o build/json_messages.o build/rs_msg_utils.o build/devices_jandy.o build/packetLogger.o build/devices_pentair.o build/color_lights.o build/serialadapter.o build/aq_timer.o build/aq_scheduler.o build/web_config.o build/serial_logger.o build/mongoose.o build/simulator.o build/timespec_subtract.o build/pda.o build/pda_menu.o build/pda_aq_programmer.o build/onetouch.o build/onetouch_aq_programmer.o build/iaqtouch.o build/iaqtouch_aq_programmer.o -lpthread -lm -lsystemd
/usr/bin/ld: build/simulator.o:(.bss+0x4): multiple definition of `_commands'; build/aq_programmer.o:(.bss+0x8): first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:194: release/aqualinkd] Error 1

```

sfeakes, 2024-05-02:
This is already fixed in the current dev branch.

ballle98, 2024-05-02:
ok I don't see it in the diff https://github.com/sfeakes/AqualinkD/compare/master...PDA-Update

john-kapp, 2024-05-12:
How do I go about cloning the git branch with this fix?  I cloned the default branch as well as the PDA-Update branch and I get the ld error on both.  Thanks very much for this project!

sfeakes, 2024-05-12:
The PDA-Update is not ready for prime time yet, so best you go back to the main / default branch.

Run the below command in whatever directory contains `simulator.c` on your machine, That will modify the file to how it should be.
```sed -i 's/_commands/_sim_commands/g' simulator.c```

john-kapp, 2024-05-12:
The sed command worked.  Thank you.

ballle98, 2024-05-21:
fixed with 675b0f5

closed 2024-05-21 by ballle98