aqualinkd/AqualinkD #290: "Notes on Aqualink PDA only (non RS panels)" link goes nowhere

ixobelle, 2024-05-08:
I'm interested in this project, as I have a "PDA P4 Only Rev 7.2" board currently. 

There is a link on this page: https://github.com/sfeakes/AqualinkD/wiki/#PDA

That says "It is recomended that you upgrade your PDA only chip to a RS & PDA chip in your control panel, here is information to do that: [Upgrade PDA to RS](https://github.com/sfeakes/AqualinkD/wiki/Upgrading-Jandy-Aqualink-PDA-to-RS-panel)", but that link just leads back to the table of contents. 

Is this PDA Only board supported, or are there instructions I'm missing? Thanks so much!

[image]

sfeakes, 2024-05-08:
That board will work with this project, but with limited functionality. Please read the wiki on pda limitations.  The chip upgrade details had to be removed from the site, so that’s no longer an option published here. Sorry for the confusion, that’s a link I need to remove.

ixobelle, 2024-05-08:
no worries, thanks for the ultra quick reply! I see that there's a bit of lag in using "PDA only" functions, but should still be quicker than "go find the remote, power it up, wait 10 sec, fumble around just to see how warm the water is or turn on waterfalls, etc".

I'll order parts tonight and give it a go, no harm in trying <3

ixobelle, 2024-05-08:
closing issue, thanks again

closed 2024-05-08 by ixobelle