aqualinkd/AqualinkD #295: Aqualink daemon can't use extended_device_id in PDA mode

ballle98, 2024-05-21:
I have a newer PDA panel that has some support.  If I patch out this check below I start seeing iAQ Touch interactions.  I was going to look into it and see if I could make it more functional.

@sfeakes have you done any additional work on this or do you have plans around this?

```
diff --git a/aqualinkd.c b/aqualinkd.c
index 79fd81a..12aa0ad 100644
--- a/aqualinkd.c
+++ b/aqualinkd.c
@@ -1606,12 +1606,12 @@ void main_loop()
 #ifdef AQ_PDA
   if (isPDA_PANEL) {
     init_pda(&_aqualink_data, &_aqconfig_);
-    if (_aqconfig_.extended_device_id != 0x00)
-    {
-      LOG(AQUA_LOG,LOG_ERR, "Aqualink daemon can't use extended_device_id in PDA mode, ignoring value '0x%02hhx' from cfg\n",_aqconfig_.extended_device_id);
-      _aqconfig_.extended_device_id = 0x00;
-      _aqconfig_.extended_device_id_programming = false;
-    }
+//    if (_aqconfig_.extended_device_id != 0x00)
+//    {
+//      LOG(AQUA_LOG,LOG_ERR, "Aqualink daemon can't use extended_device_id in PDA mode, ignoring value '0x%02hhx' from cfg\n",_aqconfig_.extended_device_id);
+//      _aqconfig_.extended_device_id = 0x00;
+//      _aqconfig_.extended_device_id_programming = false;
+//    }
   }
 #endif

```

```
15:55:49.890 Debug:   iAQ Touch: Turning IAQ SEND on
15:55:49.891 Info:    iAQ Touch: Page: HOME | 0x01
15:55:49.891 Info:    iAQ Touch: Home Status page 00| --
15:55:49.891 Info:    iAQ Touch: Home Status page 01| 95  
15:55:49.891 Info:    iAQ Touch: Home Status page 04| Pool Temp
15:55:49.891 Info:    iAQ Touch: Home Status page 05| Air Temp
15:55:49.891 Debug:   iAQ Touch: Air Temp set to 95
15:55:49.891 Debug:   iAQ Touch: Pool Temp set to 95
15:55:49.891 Info:    iAQ Touch: Button 00|          FILTER PUMP  | type=0x08 | state=0x00 | unknown=0x00 | keycode=0x11
15:55:49.891 Info:    iAQ Touch: Button 01|                SPA    | type=0x08 | state=0x00 | unknown=0x00 | keycode=0x12
15:55:49.891 Info:    iAQ Touch: Button 02|            POOL HEAT  | type=0x05 | state=0x00 | unknown=0x00 | keycode=0x13
15:55:49.891 Info:    iAQ Touch: Button 03|             SPA HEAT  | type=0x05 | state=0x00 | unknown=0x00 | keycode=0x14
15:55:49.891 Info:    iAQ Touch: Button 04|            CLEAN- ER  | type=0x01 | state=0x00 | unknown=0x00 | keycode=0x15
15:55:49.891 Info:    iAQ Touch: Button 05|           POOL LIGHT  | type=0x07 | state=0x00 | unknown=0x00 | keycode=0x16
15:55:49.891 Info:    iAQ Touch: Button 06|            SPA LIGHT  | type=0x07 | state=0x00 | unknown=0x00 | keycode=0x17

```

sfeakes, 2024-05-21:
@ballle98   Yes, their is a lot of work done around using IAQ touch for PDA panels that support that protocol. I basically decided it was best to move the whole PDA support to the IAQ protocol.  That way newer PDA panels that do NOT have iAqualinkTouch devices connected can use the IAQ for AqualinkD and PDA also the Jandy PDA remote and not have conflicts.  It's all in the latest dev branch.   set device id to 33 and panel type to a PDA panel, and it'll start using IAQ touch protocol for PDA.   There is still a lot of errors / work that needs to be complete though.
```
device_id=0x33
panel_type = PD-8 Combo
```

BTW, the latest Dev branch also has Home Assistant discovery included. ( I think you're using HA now ).  I'll be posting an update soon.
```
mqtt_hassio_discover_topic = homeassistant
```
These will also be needed to test that HA
```
force_SWG = yes
force_PS_setpoints = yes
force_Frzprotect_setpoints = yes
convert_mqtt_temp_to_c = no
```

ballle98, 2024-06-13:
OK the output from serial logger and comments in aqualinkd.conf pushed me down a bad direction.  I had set up as PDA with itouch instead of itouch without PDA like below.  Setting device_id=0x33 works

Note: There are several things broken in IA touch (heater setpoints are not working, etc.)

```
device_id=0x60
panel_type_pda = yes

# The ID for extended settings to allow for faster programming
# VARIABLE SPEED PUMP are only supported with this option.
# Do not enable this if you don't use either, you'll just waste memory and cpu cycles
# Valid ID's are 0x40, 0x41, 0x42 & 0x43. for ONE Touch
# Valid ID's are 0x30, 0x31, 0x32 & 0x33. for Aqualink Touch
extended_device_id=0x33

# If you have extended_device_id set, then you can also use that ID for programming some features.
# This means that you can turn things on/off while AqualinkD is programming certian features.
# If you are using Aqualink Touch protocol for extended_device_id then this is highly recomended
# as it will speed up programming substantially. if One Touch it's 50/50.  
extended_device_id_programming = yes

```

sfeakes, 2024-06-13:
Using 0x33 as the main ID, is not ready for prime time, (message above), hence it’s not documented in wiki or config.  There is a lot broken / not fully tested in that mode.

Using 0x33 as the extended ID, will be ignored if in PDA mode / pda panel.
(I do need to update documentation, as it was only recently I was made aware that some PDA panels do actually support 1 iAqualink touch protocol at ID 0x33)

the future for PDA panels that can support 0x33 (rev 6 or newer), will be to use 0x33 as the main ID and drop 0x60 all together.  But that’s taken a backseat at the moment while I refactor some other code to make that easier.

closed 2025-11-19 by github-actions[bot]