aqualinkd/AqualinkD #300: Got working with EW-11 wireless device!

jjarboe01, 2024-05-25:
So, for ease of use, had to make the Dockerfile pull the aqualinkd-docker.cmd and aqualinkd.conf from my local repo for easier builds, and added at the bottom of the aqualinkd-docker.cmd the following:
  echo "Starting SOCAT port binding....."
  socat -d -d pty,link=/dev/tty.Pool2,raw TCP:192.168.99.248:8899 &
  echo "Sleeping for SOCAT start....."
  sleep 5s
  echo "Starting aqualinkd......"
  /usr/local/bin/aqualinkd -d -c $AQUA_CONF

I also had to add the socat package into the docker image by editing the apt-get install line to include socat.

All of this is working with local tty.Pool port, but seeing some odd behavior on the Jandy ePump 1 - 

  Warning: RS Serial: Serial read bad Jandy checksum, ignoring
  Warning: RS Serial:  Read BAD PACKET To 0x78 of type    ePump set RPM | HEX: 0x10|0x02|0x78|0x44|0x80|0xdf|0xfb|0xff|0xff|0xff|0x10|0x02|0x78|0x41|0xcb|0x10|0x03|
  Notice:  JandyDvce: Jandy Pump found at ID 0x78 with WATTS 532, but not configured, information ignored!
  Warning: iAQ Touch: Got pump message 'Jandy ePUMP   1' but can't find pump at index 1
  Warning: iAQ Touch: Got pump message '    RPM: 2150' but can't find pump
  Warning: iAQ Touch: Got pump message '  Watts: 532' but can't find pump
  Notice:  JandyDvce: Jandy Pump found at ID 0x78 with RPM 2150, but not configured, information ignored!
  Notice:  JandyDvce: Jandy Pump found at ID 0x78 with WATTS 533, but not configured, information ignored!

I noticed that the pump cannot be set at address 0x78 but in my setup it appears the pump is addressed there.  Not sure if that was my idiot pool builder company, or if this is something new.  

Also not sure of the ?Read BAD PACKET" messages that I included above, it could be latency to the RS485 wifi adapter.  I am using this one - https://www.amazon.com/dp/B0C7RLWLF5?psc=1&ref=ppx_yo2ov_dt_b_product_details.  I don't have an option to run hard-wired right now simply because of where the controls are located at my house, and a Pi won't see wifi where its at and in the metal box.  I was able to use an external wifi antenna with it.  

So far, this is an awesome project and has taught me a TON about Docker.  Thank you for your efforts, and looking forward to getting is pump issue hashed out and working, and your tweaks for this to be a Home Assistant add-on!  Amazing work and I will be promoting this repo to many tech friends with Jandy pools!

sfeakes, 2024-05-26:
On the pump, have you added it to aqualinkd.conf?
Assuming it's the filter pump, you should have an entry like this
```button_01_pumpID=0x78```

I don't believe that bacd packet is from a latency issue, but hard to tell.  With low latency, I'd expect AqualinkD to be late in replying to a packet, and that will confuse the control panel (which may sometimes make it send a bad packet). So without seeing the chain of packets before and after that bad packet, I couldn't really tell.  AqualinkD has a build in timer, as you can't be too quick or too slow in replying, but since it *may* be getting packets late due to the latency, you should probably turn the timer off.
``` rs485_frame_delay = 0```.
(the default is 4 = 40ms, 0 is off).  But you may need to play with 1,2,3 just to see if anything changes.

BTW, it's awesome you got the EW-11 working, I'll get one and look at getting AqualinkD to read directly rather than using port bindings.  In the meantime, I'll look into modifying aqualinkd-docker.cmd so it can run custom commands.

jjarboe01, 2024-05-26:
Its a pretty basic controller setup really, Auqapure, Jandy ePump, Pentair LED lighting.  It does have solar heating, but that appears to just be an extra temperature probe and valve control wire, nothing on the RS485 bus.  Honestly not even sure if the lighting using the bus because it changes colors by flashing the relay on and off in a sequence to set lighting.  I do have a "high speed mode" for the pump they configured for when I'm cleaning the pool, but don't figure that would be a different device address.  Only other thing looking at the wiring is a iAquaLink2 wifi module, wondering if that would be whats at that address?  

The web based app is decent but they don't expose the SWG status to third party API's and refuse to put it on their roadmap because "the salt PPM is not that accurate", but I need to be able to see the status to know if my cell is having issues.  My plan is to have an automation with Home Assistant send me a notification was soon as the cell is having an issue.

Anyway, thank you for all the work on this project, it is truly amazing and I'm excited to get this where I want it finally, because I was looking at stand-along salt generators that I really didn't want to spend the money on.

sfeakes, 2024-05-26:
I think you are right, the 0xa3 is one of the address the iAqualink2 uses.  I haven't managed to get my hands on one of those to debug it yet.

The reason they won't expose the SWG status, is probably because they can't in the way they have architected the iAqualink the system.  The only reason AqualinkD can do this is because it reads every message and decodes it.  iAqualink will only read what the control panel tell / sends it.  And the control panel actively hides things from you. One example of this :-
When a VSP looses connection with the control panel, the panel will set the SWG to 0. This makes sense, as it saves the Cell from killing itself if there is indeed no flow.  Problem is the panel will loose connection with certain VSP somewhat frequently due to how Jandy implemented the RS protocol (kind-a archaic). I can see it's a bug in their firmware, but they haven't fixed it in years.  So rather than tell any device (like iAqualik / keypad etc) the SWG is at 0, it simply hides this from you for a few minutes, and if the VSP comes back online, all is good, it it doesn't then it'll finally tell you. 
Also as I'm sure you are aware, The control panel will also only tell you of a "general fault" on a SWG, and not the specific SWG fault.  So unless Jandy upgrade the firmware on the panel / come out with a new version, there is simply no way the iAqualink(1 or 2) can get the full status.  So it's for these reasons I'm sure they can't expose any detailed SWG information, and therefor won't (or rather can't) put it on a roadmap.

On the PPM, that's interesting. We test our water weekly, and the PPM reported by my Haywood cell is always spot on.  But I'm using a Haywood cell in Jandy mode, which is something else they try to block from their control panel.  As soon as the panel detects any non Jandy SWG it will lock out all SWG unless you modify the panel firmware.

Please keep me informed as to how the EW11 is performing.

jjarboe01, 2024-05-27:
Awesome information.  I figured it would show "General Fault", same as the web interface from the iAquaLink app.  Which is what leads me to think they can pull the information, they just don't want to, likely for the reasons you mentioned.  General Fault is good enough for me to know there is a problem with the system to go check, which is why I'm going down this whole rabbit hole to begin with. 

Ultimate goal is of the Display_Message is "General Fault", Home Assistant will send a notification to my phone to check the salt cell for issues.  Shame on me for buying a cheaper replacement when mine died last year, its on its way back to the company for the third time since buying it in May of 2023.  

If you want more debugging of the iAquaLink, I'm happy to run whatever you need and send you logs if it helps.  Let me know and I'll do what I can to hook you up!  

Still chasing down the config right now, thank you for the pointer on the pump config, I never made it to the bottom of the config file.  I think I have that sorted, and will be re-building my Docker container to test the changes with the frame delay.  I'll post back by next weekend on how that is working with the change, and if I have to toy with the value anymore.  I think a lot of folks would prefer to go that route and have your programming run on their HA server, vs buying a Pi and having to maintain the extra hardware.  

Again, many many thanks for the work you have put into this, this is a gift from the technical gods to mankind with a Jandy pool controller!

jjarboe01, 2024-06-15:
First off, the updates are AMAZING!  One thing to note if you copy the aqexec-pre.sh to the config directory, if you are running Docker on linux, you have to chmod +x the file before running docker compose up, otherwise the file has no execute permissions in the container.  Otherwise, this is working flawlessly, love the config folder idea to persist the files (may have been in previous version, but I overlooked).

EW11 is working great within, and now that I have a new working salt cell, this project is great.  Still have some things to debug from the serial bus on my unit, and it may be because I don't have the config file set perfectly.  I will play with that more tonight now that the pool systems are all working.

Also to note, when I was in the panel config in the iAqualink app, I accidently hit the help button and noticed my panel is RS-4 Only model, the serial logger does not show a control panel model or version.  Mine is the RS-4 Only Rev T.2, not sure if that matters or helps to add panels for your config.

jjarboe01, 2024-06-15:
For what it's worth, got this working as a self-contained add-on for Home Assistant.  Posted all of the code used on my github, of course giving you credit for the base program.  https://github.com/jjarboe01/AqualinkD-Docker-HA

If you want to incorporate my code for a pre-packaged add-on for Home Assistant, let me know and I will move my repo to private!

sfeakes, 2024-06-16:
Glad you like the changes.   I didn't even know they made an RS4 panel, (or I forgot).  Your spa/pool heater buttons will almost certainly be incorrect in the config. My guess is that Button_06 would be Pool Heater and Button_07 Spa heater (or Heat 1 & Heat 2 since it sounds like you don't have a dual panel).

That's an interesting idea to use HASSIO to run AqualinkD docker, I'd never thought about doing that before.  I'll probably add it to the documentation, and link to your repo.

sts098, 2024-07-24:
I am not sure where the best place to comment on this is.  I am looking at splitting my install across a raspberry pi and a vm on my VSphere server (run a small business with website, accounting, crm, etc on several VSphere 8 servers).  I have a Jandy pda-ps8 card with aqualink 2 for web access.  I just installed a raspberry pi 4 with atlas card and ph,temp, and Orp sensors.  Only running atlas iot software for now.  Collecting data on home assistant via Mqtt.  

Thought I might set up nodejs-pool controller on a vm to offload some of rp4 load.  After reading this, I believe I can move Aqualinkd to the vm as well.   This would only place relay equipment manager on the raspberry pi.  Has the code for the Ew-11 made it to main repo.  

Any advice on how to proceed? Can I simply connect the ew-11 and read/write data to the aqualink system?

sfeakes, 2024-07-25:
Yes you can run AqualinkD on a VM, it’s actually developed on a VM. With Ew-11 you can simply connect that to the RS485 bus and have AqualnkD on the esxi server with no other hardware required.  A PDA panel requires a LOT more communication on the RS485 bus to get even the simplest of things done, is a lot slower, and timing of commands is imperative.  No one has tested EW-11 with a PDA only panel, so this might be your issue.  The slower speed of the PDA will be an advantage, but the more precise timing might be an issue.

sts098, 2024-07-31:
I am still waiting on the delivery of the EW-11.  I will let you know how I make out.  I began installing necessary components.  It looks like jjarboe01 is using SOCAT and port binding.  Also, sfreakes was hoping to get it to read directly.  Has this been completed?  Is there any instructions on how to do this?

sfeakes, 2024-08-06:
> I am still waiting on the delivery of the EW-11. I will let you know how I make out. I began installing necessary components. It looks like jjarboe01 is using SOCAT and port binding. Also, sfreakes [...]

It will be a bit of time before AqualinkD talks directly to EW-11, since the SOCAT works and is not a massive overhead it's low priority.  SOCAT is part of the docker image, and script to start is included.  So it's super simple to implement.

bcre3306, 2024-08-15:
I'm trying to get this working as I just got a EW-11 but I'm getting this error with serial_logger:

Error:   RS Serial: Unable to open port: tty.Pool, error 16
Error:   RS Serial: Unable to open port: tty.Pool
Device or resource busy: tty.Pool

Any suggestions?  Doesn't look like a permission issue from what I can tell.  The only changes I made on the EW-11 was getting it on my network wifi which works fine, I can ping it and get to the config page on it.

bcre3306, 2024-08-15:
I did, I changed it to 192.168.1.15X which is the internal IP it's on

sfeakes, 2024-08-15:
> You have to update the aqexec-pre.sh file to update the IP of the EW-11 device.

Yes.

bcre3306, 2024-08-15:
I did this, see the logs below
Starting SOCAT port binding.....
Sleeping for SOCAT start.....
2024/08/15 22:07:44 socat[12] N PTY is /dev/pts/0
2024/08/15 22:07:44 socat[12] N opening connection to AF=2 192.168.1.150:8899
2024/08/15 22:07:44 socat[12] N successfully connected from local address AF=2 192.168.1.239:59018
2024/08/15 22:07:44 socat[12] N starting data transfer loop with FDs [5,5] and [7,7]
Starting periodic command scheduler: cron.

bcre3306, 2024-08-16:
I tried changing a few things, it was sharing a 485 connection on the RS panel which I changed to a dedicated one but that didn't seem to change anything.  I feel like the EW-11 is possibly configured wrong but I'm not seeing how you have the serial port and communications tabs set in the EW-11.  Can you post those so that I see if that helps.  I tried changing the port on the EW-11 and the startup script to see if that was it but it's not.  Also I can see on the EW11 interface that it's connecting to the AqualinkD as it shows the connection but basically no traffic is flowing back and forth.

sfeakes, 2024-08-17:
There is some information in setting up the EW-11 with Pentair controller in the below link.  It should be the same for Jandy.

https://github.com/tagyoureit/nodejs-poolController/wiki/Elfin-EW11-RS485%E2%80%90WiFi-setup

bcre3306, 2024-08-17:
ok now we have communication with the panel!  I'll work on getting the rest of it setup now.  If you add this to the wiki I would suggest you add the setting for the EW11.  Thank you again!

sfeakes, 2024-08-17:
> ok now we have communication with the panel! I'll work on getting the rest of it setup now. If you add this to the wiki I would suggest you add the setting for the EW11. Thank you again!

If you can document the EW11 config, I'd be glad to add it to the wiki.  Without having an EW11, I have no idea what the steps are.

bcre3306, 2024-08-18:
Sure thing! I attached the 3 screenshots of the configuration that are important.  One thing to note is that you need to go to the Others tab after you make all the changes and reboot(it's listed as restart).

Hope this helps!!

I now have another issue but I'll start a new thread for that because I believe it's not related to the EW-11, if it ends up being that then I'll link it back to this.

Thanks again
[EW11 - Communications Settings.pdf](https://github.com/user-attachments/files/16649532/EW11.-.Communications.Settings.pdf)
[EW11-Serial Port Settings.pdf](https://github.com/user-attachments/files/16649533/EW11-Serial.Port.Settings.pdf)
[EW11-System Settings.pdf](https://github.com/user-attachments/files/16649534/EW11-System.Settings.pdf)

Note 

When you first power up the unit you need to connect to it via the SSID that it broadcasts which is going to be EW11_XXXX(the XXXX being a random generated letters/numbers).  Once connected then you can log into the management page at 10.10.100.254 with the username and password admin/admin.  Once you get it setup on your home network then you have to get on the router to see what IP it's assigned then you can log back into the webpage with the new local IP address(once you connect your computer/phone back onto the home network).

The "BSC" in STA SSID is my home SSID so it should show yours after you enter it or select it.  You can type in your home network SSID or click the scan button to pick it from the list.

tunatoksoz, 2024-08-31:
Are you able to get this working reliably?

I have both EW11 and DX-CP24 RS485 (they look very similar lol).

While i can read status, writing feels very @bcre3306 flaky. It jumps around, i may have to click filer on 4-5 times and one wil stick.
At first, when i tried with EW11, i thought it was a latency issue. now also tried with bluetooth, result is the same.
thoughts?

bcre3306, 2024-08-31:
I got another EW11 with the actual external antenna and my packet drops stopped but seem to still have the random turning on of either spa or blower(even though it seems more spa not and not the blower) but not sure if it’s related to the EW11 or the panel version.  Shaun is looking into it, I’m sure he will keep us all posted.

sjames99, 2025-04-11:
Anyone been able to get this to work?  I can get it to work flawlessly for about 5-10 minutes before the serial block causes the port to be overwhelmed and the connection is lost.

jkosharek, 2025-04-18:
AquaLinkD running in Debian Bare metal instal VM and EW-11 working connected to my RS-15 Dual and can read from the panel. I am using this command to bring up the serial:
socat -d -d pty,link=/dev/ttyEW11,raw,ignoreeof TCP4:10.11.12.15:8899,ignoreeof

Is the the right way? How do I get the socat command to run after a reboot?

jkosharek, 2025-04-18:
So what I did was to create a service for the EW11 (/etc/systemd/system/aqualinkd-ew11.service)
```
# Content of /etc/systemd/system/aqualinkd-ew11.service
[Unit]
Description=AquaLinkD EW11 RS-485 Bridge
After=network.target

[Service]
Type=Simple
StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=aqualinkd-ew11-rs-485
ExecStart=socat -d -d pty,link=/dev/ttyEW11,raw,ignoreeof TCP4:10.11.12.15:8899,ignoreeof
Restart=always

[Install]
WantedBy=multi-user.target
```
```
systemctl daemon-reload
systemctl enable aqualinkd-ew11.service
```
Added 'Requires=aqualinkd-ew11.service' in '/etc/systemd/system/aqualinkd.service'. 

Seems to be working