aqualinkd/AqualinkD #305: iaqtouch_aq_programmer.c:942:11: error: invalid type argument of unary '*'

ballle98, 2024-05-29:
not sure why this is only seen during cross compile should either remove the dereference "*" or change "->" to "."

https://github.com/sfeakes/AqualinkD/blame/322924617df9bd39b2a152fd4aae6ab1ea69db2a/iaqtouch_aq_programmer.c#L942

```
arm-linux-gnueabihf-gcc -Wall -O3  -D AQ_PDA -D AQ_ONETOUCH -D AQ_IAQTOUCH -D AQ_RS16 -D AQ_MANAGER -D MG_ENABLE_HTTP_SSI=0 -D MG_ENABLE_DIRECTORY_LISTING=0 -D MG_ENABLE_HTTP_CGI=0 -DGIT_HASH="\"fdfa8b9\""  -c -o build/iaqtouch_aq_programmer.o iaqtouch_aq_programmer.c
iaqtouch_aq_programmer.c: In function 'get_aqualink_iaqtouch_setpoints':
iaqtouch_aq_programmer.c:942:11: error: invalid type argument of unary '*' (have 'int')
  942 |       if (*button->name[8] == 'C') {
      |           ^~~~~~~~~~~~~~~~
make: *** [build/iaqtouch_aq_programmer.o] Error 1

```
Here is a patch

```
@@ -939,7 +939,7 @@ void *get_aqualink_iaqtouch_setpoints( void *ptr )

     if (button != NULL) {
       LOG(IAQT_LOG,LOG_NOTICE, "Temperature units are '%s'\n",button->name);
-      if (*button->name[8] == 'C') {
+      if (button->name[8] == 'C') {
         aq_data->temp_units = CELSIUS;
       } else {
         aq_data->temp_units = FAHRENHEIT;
```

ballle98, 2024-05-31:
This makes no sense.  I tried different test code to get to the bottom of it and the compiler thinks name is pointer to char pointer instead of char pointer.  

This is a test 

```
    if (button != NULL) {
      LOG(IAQT_LOG,LOG_NOTICE, "Temperature units are '%s'\n",button->name);
      struct iaqt_page_button tButt;
      char *bname = tButt.name;
      char cTemp = bname[8];
      if (cTemp == 'C') {
        aq_data->temp_units = CELSIUS;
      } else {
        aq_data->temp_units = FAHRENHEIT;
      }
    }

```

This is the warnings from compiler

```
iaqtouch_aq_programmer.c: In function ‘get_aqualink_iaqtouch_setpoints’:
iaqtouch_aq_programmer.c:943:21: warning: initialization of ‘char *’ from incompatible pointer type ‘char **’ [-Wincompatible-pointer-types]
  943 |       char *bname = tButt.name;
      |                     ^~~~~
iaqtouch_aq_programmer.c:944:12: warning: ‘*((void *)&tButt+8)’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  944 |       char cTemp = bname[8];
      |            ^~~~~

```

The structure is defined as char array

```
struct iaqt_page_button {
  char name[IAQT_MSGLEN];
  unsigned char type;    // 0x01 box, 0x08 icon wirlpool, 0x0b icon heater, 0x01 icon (main page), 0x07 light
  unsigned char state;
  unsigned char keycode;
  unsigned char unknownByte;
};

```

sfeakes, 2024-05-31:
I couldn't make any sense of that either.  I tried a bunch of different GCC versions and couldn't replicate the error.
But, since you are using armhf cross compiler, I really couldn't find / get a version from the net that worked well, and certainly not for older installs (stretch). So I ended up using creating my own docker to support making release binaries for AqualinkD.  This will compile for maximum compatibility
# armhf is 32 bit armv6l (armhf) stretch and newer - work on all Pi's running 32bit (Pi0 to Pi4)
# arm64 is 64 bit aarch64 buster and newer - work on Pi3/Pi4/2w running 64bit os

https://github.com/sfeakes/AqualinkD-Docker/blob/main/Dockerfile.releaseBinaries

Since it compiles  GCC / LibC & BinUtils, it'll take some time to make the container the first time.

ballle98, 2024-06-10:
I figured it out in iaqt_page_button name is defined as an array of char pointers instead of a char array
I had changed this on my fork as part of an effort to clean up code scanning issues https://github.com/ballle98/AqualinkD/issues/85

```
index 61deb7e..ff28691 100644
--- a/iaqtouch.h
+++ b/iaqtouch.h
@@ -5,7 +5,7 @@
 #define IAQT_MSGLEN 21
 
 struct iaqt_page_button {
-  char *name[IAQT_MSGLEN];
+  char name[IAQT_MSGLEN];
   unsigned char type;    // 0x01 box, 0x08 icon wirlpool, 0x0b icon heater, 0x01 icon (main page), 0x07 light
   unsigned char state;
   unsigned char keycode;
diff --git a/iaqtouch_aq_programmer.c b/iaqtouch_aq_programmer.c
index 86bff4a..8952796 100644
```

sfeakes, 2024-06-11:
Thanks very much.  It's always the stupid things that take the longest.

PR #312 (2024-06-11): ballle98/AqualinkD#93: iaqt_page_button.name should be char[] not *char[]
fix #305
closed 2024-06-11 by sfeakes