aqualinkd/AqualinkD #324: "can't find pump" message from variable speed pump output

ssorkin, 2024-07-23:
I have a Jandy RS board (Yg / 4.3.13 firmware), that was the upgrade for the iAqualink 3 (since I previously had a PPD PDA chip). I also have a Century Variable Speed Motor (that acts like an ePump on RS485), a TruSense chemistry sensor and an AquaPure. All communication is via the Elfin EW-11 (which works great).

AqualinkD currently complains about not being able to find the pump, presumably because this board doesn't report one of the well known pump strings (in the `passDeviceStatusPage` function in `iaqtouch.c`). I uncommented the log line and see the following output as interpreted over the serial bus:

`Notice:  iAQ Touch: Passing message 00| Filter Pump
Notice:  iAQ Touch: Passing message 01| Filter Pump
Notice:  iAQ Touch: Passing message 02|     RPM: 1000
Warning: iAQ Touch: Got pump message '    RPM: 1000' but can't find pump
Notice:  iAQ Touch: Passing message 03|   Watts: 0
Warning: iAQ Touch: Got pump message '  Watts: 0' but can't find pump
Notice:  iAQ Touch: Passing message 04| ORP:670 pH:7.6
Notice:  iAQ Touch: Passing message 05| #1 AquaPure
Notice:  iAQ Touch: Passing message 06| 35%
Notice:  iAQ Touch: Passing message 07| Salt 3300 PPM`

Is it obligatory to name the pump as desired by this function or should it be able to interpret the label (which is the default).

Note that my configuration is:

`button_01_label = Filter Pump
button_01_pumpID = 0x78
button_01_pumpIndex = 1
button_01_pumpType = Jandy ePUMP`

sfeakes, 2024-07-24:
On the messages part, usually you would see
`Filter Pump

Pump 1

RPM: xxx`

And the 1 needs to match the `button_01_pumpIndex = 1` so that’s why AqualinkD is complaining.  But since you have `button_01_pumpID = 0x78` aqualinkd should be getting all the pump info from the RS buss directly.  So it’s probably just an annoying warning message at this point.

it’ll be a few weeks before I’m infront of a computer where I can make any changes to overcome this.  You might be best simply commenting out the warning message for the moment.

ssorkin, 2024-07-25:
I don't really care about the warning messages; my issue here is that the bus monitoring code doesn't seem to interpret the pump traffic and hence i don't see the RPM/power without modifying the `passDeviceStatusPage` code. I can dig into the bus parsing code and enable some debug logging to see what's going on there...

ssorkin, 2024-07-25:
the bus parsing code is working fine, it seems. the data comes through on the websocket, but is discarded by the front end controller because status=-1. if i disable that check, the rpm/watts are displayed. i'll take a look as to why the status is marked as -1.

sfeakes, 2024-07-25:
-1 means AqualinkD thinks the pump is off.  That is probably because it’s not getting all the messages, like Watts and/or catching the control panel message with pump number.  So UI is doing the wright thing by not displaying pump info because status is off.   It’s a use case that will probably need a number of different changes to be made in jandy_devices.c, iaqtouch.c & oncetouch.c.    If you want to look some more, rather than looking at the websocket, it might be easier to look at the JSON directly.   Http://AqualinkD.ip/devices & http://AqualinkD.ip/status.

ssorkin, 2024-07-26:
I see where the `aqdata` structure is being updated with the appropriate RPM and Watts in `jandy_devices.c` by means of watching the bus, but I don't see where the status would ever be set to `PS_OK`. In `iaqtouch.c` it can only possibly happen in `passDeviceStatusPage` which doesn't work for me since the pump isn't named in the very particular way expected. it's unclear to me if we're calling into `onetouch.c` on my system, so it will not be `PS_OK` there either.

ssorkin, 2024-07-27:
on a separate note (and i can create another issue for this once i understand what's happening), my aquapure gets in a completely stuck state when i run aqualinkd, such that i have to power cycle the RS to be able to communicate with it. have you seen this happen?

sfeakes, 2024-07-27:
Not with your setup.  V2.3.7 fixed an issue with Pentair VSP and SWG, but I’ve never come across one with Jandy VSP and SWG.  This could also be EW-11 issues, as soon as their are any latency / protocol problems it’s the SWG that usually complains / craps out first.

ssorkin, 2024-07-28:
The interesting part here is that power cycling the SWG alone doesn't fix it. The RS itself has to be cycled to get it communicating with the SWG.

sfeakes, 2024-07-28:
It’s probably the control panel that’s causing the SWG issue.   If you haven’t tried 2.3.7 it’s worth a shot

ssorkin, 2024-07-28:
I'm on 2.3.8 (built from source from Github).

sfeakes, 2024-07-28:
2.3.8 is currently in dev, so not stable release.  That wont be the cause of SWG issue, but may be on the pump display issue.

sfeakes, 2024-08-11:
Can you provide some more detailed debug logs, so I can take a look at this in more detail and find a way to overcome this specific setup.

Go to aqualinkd's management page.  `http://aqualinkd.ip.address/aqmanager.html`

- (Turn pump off)
- Expand the debug mask section and enable `iAQ Touch` button.
- Let that run for minute
- (Turn pump on)
- Let that run for a few minutes, to catch prime then RPM messages.
- Then expand log file section, increase to 10k lines, download and post the file here.

On a separate request, I'd love to see the messages from the TruSense chemistry sensor and decode those. But first I'll need to see the ID it's using.  Can you post the output from running the serial_logger (again you can do that from aqmanager.html).

sfeakes, 2024-09-04:
@ssorkin Release 2.3.8 is now released.  It has some changes for Panel rev Yg.  The pump should work as you can now label by name (rather than pump index as per previous panel revisions).
Add the below to your config.
`button_??_ pumpName = Filter Pump`

ssorkin, 2024-09-04:
I will try this and get you some of the data from the TrueSense