aqualinkd/AqualinkD #332: Can't see any of my equipment on the webpage besides the SWG

bcre3306, 2024-08-18:
I just got started and am using an EW11 to connect back to my home network which I have AqualinkD running in two containers, one so I can run serial_logger and one to run aqualinkd.  The reason for this is because if I kill the aqualinkD process in the container it kills my container and if I try to run serial_logger while aqualinkd is running I get errors because it's locked by aqualinkd.

With that being said I can run serial_logger and see my panel.  I attached the log for you to review.  I also can see the output of the container running aqaulinkd and also attached that log which shows me that it's communicating with the panel but like the title says I only see the SWG and the salt levels.  I see all the other changes to the names of the buttons in the webpage but they don't actually turn any of the equipment on or off.  I can see in the logs where I push the buttons and it looks like it's trying to change the state of the AUX or pump but never actually changes them.  

Any help with this would be greatly appreciated.

Thanks !
[aqualinkd.conf.txt](https://github.com/user-attachments/files/16649746/aqualinkd.conf.txt)
[Startup.txt](https://github.com/user-attachments/files/16649747/Startup.txt)
[serial_logger.txt](https://github.com/user-attachments/files/16649748/serial_logger.txt)

sfeakes, 2024-08-18:
On the first issue, there are two options.  

1. Run serial_logger from the AqualinkD management console.  `http://aqualinkd-ip/aqmanager.html`
2. Run the container manually, and start either AqualinkD or serial_logger. Something like `sudo docker run -it aqualinkd /bin/bash`. Command line but will depend a lot on your setup.

For the other issue, you need to get rid of the checksum errors.  First make sure all the wiring is correct, double / triple check that as that's 99% where those issues lie on normal system. (ie without the EW11).  Also while you are getting it stable it would be good to reduce the packets AqualinkD needs to reply to. So comment out `extended_device_id = 0x30` in your config.
Looking at `Startup.txt` you didn't give AqualinkD enough time to connect to the control panel and retrieve all the settings. Not sure if you didn't wait long enough, or you didn't post all the log.  But it can take ~1 min for AqualinkD to get all the info it needs.

bcre3306, 2024-08-19:
I blew away the containers and just spun up a VM for this to get it figured out then I'll work on getting it working in a container.  I also had it wired with the iaqualink antenna so not sure if that was causing some of the checksum errors but I have it running on its own RS485 and here is the log from it.  Thanks again
[aqualinkd.log](https://github.com/user-attachments/files/16666701/aqualinkd.log)

bcre3306, 2024-08-20:
Disregard.  The issue I was having was MQTT for some reason, as soon as I commented it out the solution is now able to control the pool via the webpage.  I need to do some more cleaning up on it and testing but seems to be working at the moment.  

On a side note it looks like more people are looking into the EW11 as a solution to communicate with aqualinkD.  I was thinking of starting a dedicated thread to this if you think it's worth it, just let me know.

bcre3306, 2024-08-20:
actually it wasn't MQTT it was the setting that @sts098 suggested in his other thread for the EW11 settings.  Changing the gap time to 10MS and buffer to 512 seemed to get me communicating correctly now!  I'll update my screenshots for the EW11 configs and post them shortly.

sfeakes, 2024-08-20:
Good idea I just started that.

https://github.com/sfeakes/AqualinkD/discussions/336

closed 2025-11-19 by github-actions[bot]