aqualinkd/AqualinkD #334: Aqualinkd.service failed because the service did not take the steps required by its unit configuration

dadoadk, 2024-08-19:
Intro: I followed the steps in the Wiki to install 
1. required dependencies, 
2. cloned the repo, 
3. without make I'm doing an install of the AqualinkD by running `sudo ~/software/AqualinkD/release/install.sh`

then I get some messages:
```
Arch is arm64, Using 64bit AqualinkD
Please enabled cron.d support, if not AqualinkD Scheduler will not work. ...
Created symlink etc/systemd/system/multi-user.target.wants/aqualinkd.service -> /etc/systemd/system/aqualinkd.service
```
at this point I edit the cron file:
`sudo nano /etc/default/cron`
and enable the -l option

now I can run the install:
`sudo ~/software/AqualinkD/release/install.sh`

and there are some messages about AqualinkD config exists, did not copy new config...
and at last a message: Starting daemon aqualinkd

to get a better status I run the command:
`sudo aqualinkd -d -c /etc/aqualinkd.conf`

which leads me to the issue I'm having:
```
Notice: NetService:Starting network services thread
Error: RS Serial: Unable to open port: /dev/ttyUSB0, error 2
Error: AqualinkD: Error Aqualink setting serial port: /dev/ttyUSB0
Notice: NetService: Starting web server on port 80
```

to confirm, I run the command `systemctl status aqualinkd.service` 
I see `Active: failed (Result: exit-code)`

my questions:
1.  should the web server be running at this point? 
2.  since I don't have anything plugged into the USB0, is the software expecting something to be plugged in?
3.  when I open my browser on localhost:80, or 192.168.xx.xx:80 should there be a page already running?

 
I've read the Wiki document several times, and some expectations are that the AqualinkD service should be running before I get to plug it into my pool unit outside my house, but I'm maybe missing a point here. Please help me understand if I'm missing something.

sfeakes, 2024-08-19:
Without a connection to the Jandy control panel, AqualinkD can’t really do anything.  So it’s failing since it can’t read the serial port /dev/ttyUSB0.

sfeakes, 2024-08-21:
That is how aqualinkd works, but since it gets the information for the UI from the panel itself, there isn’t really any information to show in the UI.  Ie it has no idea is Pool Heater is on or off / what temp it’s set to or even if it exists and can change any setting on the heater.  So you can actually catch Aqualinkd UI before it terminates itself, but the tiles are empty.

dadoadk, 2024-08-21:
By that explanation I should be able to open my browser on localhost:80, or 192.168.86.xx:80 and see an empty page. However, now I'm seeing:
`This site can’t be reached 192.168.86.xx took too long to respond.`

I'll give it a try once I get the USB2RS485 dongle.

sfeakes, 2024-08-22:
AqualinkD will start all the network interfaces, then open serial port and connect to panel. If it can’t open a serial port it will terminate, if it opens the serial port and doesn’t see or connect to a panel, it will terminate.
If you connect to the web interface after it starts the web (& MQTT) services, and before it terminates, you will see a blank page.  The time it takes to terminate after it’s started the network interfaces will depend on serial port being available, finding an RS485 adapter, seeing some data on that serial connection.  It could terminate almost immediately or after a few seconds.

dadoadk, 2024-08-26:
I've got it working. I was able to create a schedule, tested it out for a couple days. The communication with pump works great. I'm not getting lights to work, even lights on schedule don't turn on. Is there something I can do to troubleshoot the lights?

sfeakes, 2024-08-27:
I was just tuning some tests, and it looks like there is a bug with color light and certain panel versions. So I'm assuming this is a color light.  If so, I will post a fix as soon as I figure out the problem.
For the moment, you can 

1. Use one of the simulators to turn on the light
2. Delete the color light from the Jandy control panel, and let AqualinkD program it.

Sorry I realize neither is optimal.

sfeakes, 2024-09-04:
@dadoadk version 2.3.8 has a fix for color lights.  Please make sure you have this line uncommented in the config 'rssa_device_id=0x48 '

dadoadk, 2024-09-09:
@sfeakes the pool light works now. I followed your instructions, installed the new 2.3.8 version and changed the config. I've been testing it over the weekend with the scheduler. I'm not sure if the color can be changed from the UI. All colors turn on at the same time.

sfeakes, 2024-09-09:
In the web UI a long button press on the tile will give you the option to program a different color.

closed 2025-01-29 by sfeakes