aqualinkd/AqualinkD #337: How to control JVA assigned to Solar and Cleaner

bcre3306, 2024-08-22:
Hi

So I have two issues, first being I have button 08 assigned to "bubblers" in my aqualinkd.conf which looks like it's actually assigned to the Solar button which I believe I have the solar button assigned to Aux_V27 which is the JVA needed to turn the bubblers on(note I don't use a solar heater so I stole the JVA plugs for two other use cases).  When I click on the bubblers button in aqualinkd it states it's turning on the solar_heater which triggers it to turn on "Extra Aux" instead of "Aux_V27".  I have the JVA correctly assigned in the iaqualink app solar but can't seem to get it to pull the correct AUX.  

Is there anyway to assign and AUX_V to a button?  Thats really what I'm trying to accomplish

The second issue I have is similar to the above except another JVA assigned to Cleaner which I can't get Cleaner or Waterfall to show up on button 09 or anything above button 08.  Assuming if I can get the above issue that this is just another config issue but was hoping someone could comment

sfeakes, 2024-08-22:
Can you describe you setup a little better, ie control panel type / version etc.
To my knowledge you can’t reassign a button, you can however assign a JVA to a button.
Extra-aux and solar-heater are the same button, it just depends on the panel version as what its default label is.
If you can’t get anything more that button 8 showing up, you have the wrong panel type configured in aqualinkd (assuming you have a panel larger that RS8.  Btw if you have an RS10 or RS16 but have RS8 configured, then extra-aux or solar heater will turn on the incorrect button so this might be that problem.

bcre3306, 2024-08-22:
Hi

I have a RS4 combo which is what I have in the conf file.  I may be using wrong terms here or somewhat confused but I don't use any physical buttons on my panel as it's on the side of my house, everything I have ever used is in the Aqualink app and thats where I've always configured the JVA assignments to the AUX's.  I attached a screen of iaqaulink app to show the 8 buttons I have on the Home Screen which I can change around I believe but not sure if thats where I'm confused.....is that where the mapping should be back to in the conf file?

Thanks

sfeakes, 2024-08-23:
An RS4 combo has 4 circuits. (4 in RS4 is the # of circuits).  The circuits on RS4 are Filter Pump, Aux1, Aux2, Aux3. 
On top you can also assign the Solar Heater as a circuit as well, sometimes called Extra Aux.  Since it's a combo the Spa also has a circuit. Then you have the "Heater". Each of these are a Button on the control panel (and also in their RS protocol).
So total of 8 buttons.  I know it's confusing, but this is how Jandy have decided to label / call it.
```
Labeling for RS4 Combo
------------------------
Button_01   | Filter Pump  
Button_02   | Spa         
Button_03   | Aux 1       
Button_04   | Aux 2        
Button_05   | Aux 3        
Button_06   | Pool Heater  
Button_07   | Spa Heater   
Button_08   | Solar Heater
``` 
All of those circuits you can also assign a JVA to, and RS4 can support 4 JVA's.  So I am confused as to your Aux_V27 as even on an RS16 panel, that goes to Aux 1 to 7 and Aux B1 to B8 (20 buttons total),   No Aux_V27.

So your RS4 has 8 buttons that you can utilize.  01,02,06 & 07 are fixed, 03,04,05 & 08 you can play / assign as you like.
From your screen shot (if you haven't re-ordered things), 
```
Button_03 = Aux 1 = Spa Blower
Button_04 = Aux 2 = Spa Light
Button_05 = Aux 3 = Pool Light
Button_08 = Solar Heater = (Something listed under the other devices in screen shot , maybe Bubbler??) 
```

Very simply, look at your control panel, put it in service mode, and select each button, it will show you what device it is assigned to, and that's the config for AqualinkD. 

[https://cdn11.bigcommerce.com/s-ytojrjy0sf/images/stencil/1280x1280/products/42186/78632/JDY-301-0024__16943.1539017314.jpg?c=2](url)

bcre3306, 2024-08-23:
Thanks for the clarification!  I think my problem is the solar heater button that I have mapped to the bubbler JVA won't turn on with the heater button because it's not actually configured for solar(I don't have solar panels).  I use the solar JVA spot to turn on the bubblers which I can turn on a specific AUX_* or AUX_V* via the integration I have in HA.  I was trying to achieve this also through AqualinkD so that I could turn on Aux_V27(Bubblers) and also change the VSP to a specific RMP with one click.  In theory I can use a combination of HA and AqualinkD to accomplish this but was hoping to eliminate the HA pieces as they rely on the Jandy cloud.  

For the statement of having Aux_V27 on a RS4 unit there are two things that differentiate AUX's in this unit, Aux_* and Aux_V*.  The Aux_V* are virtual aux's that the system has(I have a new RS4(Y revision) and it creates Aux_V1-V28.  I just randomly assigned Aux_V27 and V28 to the two JVA's that I have for turning on the Waterfall JVA and Bubblers JVA.

I can send you outputs for this panel if it would help, just let me know!

sfeakes, 2024-08-23:
If you can post the outputs of the panel, and any online documentation that would be great. I have not seen this aux_v* before.  I think I know how they have integrated it, and it will make probably make for a somewhat slow and problematic integration into any automation hub.  IE I'm sure it only have a soft on/off not a physical on/off in the protocol.

Could you also post some screen shots of you "other devices on/off" page, and how you configure the Aux_V* ?

bcre3306, 2024-08-24:
Attached the pictures and startup log that you requested.
[Aqualinkd_Startup.txt](https://github.com/user-attachments/files/16734764/Aqualinkd_Startup.txt)
[image]
[image]
[image]

sfeakes, 2024-08-24:
Thanks very much for this, very helpful.  I can that "Bubblers, Extra Aux & Cleaner" are turned on/off from the usual Devices page, but how do you turn Aux V26 on/off?  Can you post a screen shot of that.
Also on the "Status", I'm assuming that you do NOT see the status of "Cleaner / Clean Mode". ie you only know that's on from the "Devices" page.  Can you check that by turning on "Bubblers, Extra Aux, Cleaner & Aux V26" and see if any are listed in the "Status" page.

I do not have access to the iAqualink 2.0 for 3.0 devices, so it's very hard for me to see the differences between iAqualink Touch protocol (that they all use), and any changes for v 2 and 3 they have made.

Would you also much going to the aqmanager.html page, and running `serial_logger`. And while the serial_logger is running change some devices on/off from the mobile app, as I'd love to see if that uses a different ID to the web interface.  This is the mobile app I am referring too.  (Then post the output from the serial_logger)
[image]

As for your desire to have 1 button that can turn on devices & set VSP RPM quickly and not through Jandy cloud, AqualinkD can do that already, but not for the devices you want to use.  So yes we can do this, but I will need some more debugging information from your system, so there may be some more requests for information if you don't mind.

BTW, from that log, I think that the bad packet you are seeing, is due to a protocol change in Board Rev Yg that has changed, and not an actual RS485 issue.

bcre3306, 2024-08-24:
Ok! Let me try to tackle your questions one at a time to make sure I get them all.

When I assign the JVA to Cleaner(Aux_V27) or Solar(Aux_V28) in the iaqualink web app it then takes the name of the  AUX labels I created in the iaqualink web app and then updates them in the devices list in the iaqualink app(not web page app, iOS app).  Basically it's just taking the two JVA slots on the panel for Solar and Cleaner and allowing me to rename them.

Nothing shows up in the status window or logs for the "bubbles" or "waterfall" when I turn it on or off via the iaqualink app.  I don't see anything in the logs indicating this either.

Attached the serial_logger here: 
[aqualinkd-2.log](https://github.com/user-attachments/files/16737320/aqualinkd-2.log)

Here is a picture of the equipment page that you requested:
[image]

Hope I captured everything you requested.  Just let me know what else you need.

sfeakes, 2024-08-25:
Thanks that’s very helpful.
I will need some more information, but I’m going to make some changes to the next release to make it easier for you to capture that information.  I’m basically adding the serial-logger command line flags to aqmanager, that way you don’t need to run the container in manual mode.

if you want to take this offline, can you shoot me an email at sfeakes@gmail.com

closed 2025-05-24 by sfeakes