aqualinkd/AqualinkD #341: Serial Log Not working

sapozd, 2024-09-05:
I have everything wired and setup. I can access the control panel, but the button presses are not doing anything.

I am attempting to use the serial logger to get info, but it just keeps hanging:
```
AqualinkD serial_logger V2.6
Notice:  RS Serial: Port /dev/ttyUSB0 low latency mode is set
Notice:  Serial Log:Logging serial information!
```

No other information is available past this point.

Could it be my connection?

This is the data I see when running sudo -s eval 'stty raw -echo < /dev/ttyUSB0; cat -vte /dev/ttyUSB0':
```
M-7EM-7M-oM-^?M-_M-^?M-^?M-^?M-^?M-'M-_M-_M-{M-}M-^?M-_M-yM-7_M-^?M-0M-^W^@M-?oM-^?M-^?M-^?M-^?M-'M-_M-yM-_M-{M-}M-^?M-^?M-YM-_M-y^@M-7M-yM-^?M-+M-^M5M-}M-kM-*M-jM-5^ZM-^BM-+M-$M-^WM-{M-^?M-^?M-^?M-_M-yg]9M-_^@M-7M-{M-^?M-_M-^?M-^?M-^?M-^?M-'M-_M-^PM-^?M-{M-^?M-}M-^?M-^?M-YM-_M-y^@[M-^?7M-xM-^W^@M-oM-{M-^?M-_M-^?M-^?M-^?M-^?M-xM-^WM-^?M-{M-^?M-}M-^?M-^?M-YM-_^@M-7Y5M-_M-y^@M-7M-{M-^?M-xM-^?M-^?M-'M-_M-yM-_M-{M-^?M-^?M-^?M-YM-_M-yM-7M-mM-IM-_M-y^@M-<M-{M-_M-^?M-^?M-^?M-^?M-'M-_M-^?M-_M-{M-^?M-^?M-^?M-YM-_M-y^@M-7M-kM-^?M-nM-^W^@M-7M-^?M-_M-^?M-^?M-HM-^?M-^?M-}M-^?M-^?M-YM-_M-y^@M-iM-^?M-EM-xM-^W^@M-oM-{M-^?M-_M-^?M-^?M-^?M-'M-_M-yM-^?M-_M-{M-}M-^?M-^?M-YM-_M-y^@M-pM-}[M-_M-y^@M-{M-^?M-^?M-^?M-^?M-^?M-'M-_M-yM-_M-{M-^?M-}M-^?M-^?M-YM-_^@M-7M-^?YM-_M-y^@M-7M-{M-^?M-_M-^?M-^?M-^?M-^?M-'M-_M-yM-^?M-_M-{M-^?M-^?M-^?M-_M-yM-7{M-uM-^W^GM-?M-^?M-_M-^?M-^?M-^?M-'M-yM-^?M-{M-^?M-^?M-^?M-_M-yyM-^?M-y^@M-oM-{M-^?M-_M-^?M-^?M-^?M-^?M-'M-xM-^WM-^?M-_M-{M-^?M-}M-^?M-Y^@M-7M-^?M-;^@M-7M-{M-^?M-^?M-^?M-^?M-'M-_M-yM-_M-^?M-}M-^?M-Y^@M-7M-9M-_M-yM-7M-<M-^?M-_M-^?M-^?M-^
```

Bugman1400, 2024-09-08:
Make sure you have AqualinkD v2.3.7 or later. Then, use the Web UI AQ Manager to get the serial logging to your Aqualink RS control panel. The AQ Manager is at 192.168.1.xxx\aqmanager.html. Then post results.

sapozd, 2024-09-08:
I'm running 2.4.0

Here's what happens when I run serial logger from the web UI.

```
Debug: NetService:JSON: web used 2324 of 5120
Debug: NetService:WS: URI Request 'devices': value -999.00
Debug: NetService:WS: Message - Key 'uri' Value 'devices'
Debug: NetService:JSON: web used 2324 of 5120
Debug: NetService:WS: URI Request 'devices': value -999.00
Debug: NetService:WS: Message - Key 'uri' Value 'devices'
Debug: NetService:JSON: web used 2324 of 5120
Debug: NetService:WS: URI Request 'devices': value -999.00
Debug: NetService:WS: Message - Key 'uri' Value 'devices'
Debug: NetService:JSON: web used 2324 of 5120
Debug: NetService:WS: URI Request 'devices': value -999.00
Debug: NetService:WS: Message - Key 'uri' Value 'devices'
Notice:NetService:Received request to run serial_logger!
Debug: NetService:WS: URI Request 'seriallogger/--/false': value 800.00
Debug: NetService:WS: Message - Key 'value' Value '800'
Debug: NetService:WS: Message - Key 'uri' Value 'seriallogger/--/false'
Debug: NetService:Started AqualinkD Manager
Debug: NetService:WS: URI Request 'aqmanager': value -999.00
Debug: NetService:WS: Message - Key 'uri' Value 'aqmanager'
Debug: NetService:++ Websocket joined
Debug: NetService:WS: URI Request 'status': value -999.00
Debug: NetService:WS: Message - Key 'uri' Value 'status'
Debug: NetService:JSON: web used 2324 of 5120
Debug: NetService:WS: URI Request 'devices': value -999.00
Debug: NetService:WS: Message - Key 'uri' Value 'devices'
Debug: NetService:++ Websocket joined
Debug: NetService:Served WEB request
Debug: NetService:API: URI Request 'dynamicconfig': value 0.00
Info:  NetService:URI request: : '/api/dynamicconfig '
Debug: NetService:Served WEB request
Info:  NetService:URI request: : '/aqualinkd.png '
Debug: NetService:Served WEB request
Info:  NetService:URI request: : '/ '
Debug: NetService:-- Websocket left
Debug: NetService:Stoped Aqualink Manager
Debug: NetService:-- Websocket left
```

sfeakes, 2024-09-08:
That's all information from AqualinkD UI making requests to the AqualinkD server. Since you have debug set and are not seeing anything from the serial port, that's your problem.

From your raw dump of the /dev/ttyUSB0 it kind-a looks like you have data + and data - round the wrong way. BUT that could also be the terminal / ssh you are using.

When you say you can access the "control panel", do you mean AqualinkD's web UI?  Or something else?

The fact that the serial_logger is not displaying anything probably means you have some issue with the connection, or you are not waiting long enough.  Try running serial_logger with the -d switch and let us know if you see anything that way.

You should probably run AqualinkD from the terminal at the moment, and posting that output here.
```
sudo systemctl stop aqualinkd
sudo /usr/local/bin/aqualinkd -d -c /etc/aqualinkd.conf
```

sapozd, 2024-09-08:
I mean the web UI, sorry.

I tried switching the wires. this is the data that is coming out of the raw dump now:

`^B^@^@^S^@^P^A^@^P^C^B^@^@^S^@^P^A^S^P^P^B^@^@^C^@^@^A^S^P^C`

however, it's still not working. and the serial logger is still not returning anything:

```
Notice:  AqualinkD: Aqualink Daemon v2.4.0
Notice:  AqualinkD: Panel set to RS-8 Combo Pool/Spa
Notice:  AqualinkD: Config log_level         = 5
Notice:  AqualinkD: Config device_id         = 0x0b
Notice:  AqualinkD: Config rssa_device_id    = 0x00
Notice:  AqualinkD: Config extra_device_id   = 0x00
Notice:  AqualinkD: Config extra_device_prog = NO
Notice:  AqualinkD: Config serial_port       = /dev/ttyUSB0
Notice:  AqualinkD: Config socket_port       = 80
Notice:  AqualinkD: Config web_directory     = /var/www/aqualinkd/
Notice:  AqualinkD: Config use_aux_labels    = NO
Notice:  AqualinkD: Config override frz prot = NO
Notice:  AqualinkD: Config mqtt server       = (null)
Notice:  AqualinkD: Config mqtt DZ sub topic = (null)
Notice:  AqualinkD: Config mqtt DZ pub topic = (null)
Notice:  AqualinkD: Config mqtt AQ topic     = (null)
Notice:  AqualinkD: Config mqtt user         = (null)
Notice:  AqualinkD: Config mqtt passwd       = (null)
Notice:  AqualinkD: Config mqtt timed update = NO
Notice:  AqualinkD: Config mqtt HA dis topic = (null)
Notice:  AqualinkD: Config mqtt ID           = aqualinkd_b827eb65
Notice:  AqualinkD: Config force SWG         = NO
Notice:  AqualinkD: Config force PS setpoint = NO
Notice:  AqualinkD: Config force Freeze Prot = NO
Notice:  AqualinkD: Config deamonize         = NO
Notice:  AqualinkD: Config enable scheduler  = YES
Notice:  AqualinkD: Config light_pgm_mode    = 0.00
Notice:  AqualinkD: Debug RS485 protocol     = NO
Notice:  AqualinkD: Debug RS485 protocol raw = NO
Notice:  AqualinkD: Display warnings in web  = NO
Notice:  AqualinkD: Keep panel time in sync  = YES
Notice:  AqualinkD: Read SWG direct          = NO
Notice:  AqualinkD: Read ePump direct        = NO
Notice:  AqualinkD: Read vsfPump direct      = NO
Notice:  AqualinkD: Read JXi heater direct   = NO
Notice:  AqualinkD: Read LX heater direct    = NO
Notice:  AqualinkD: Read Chem Feeder direct  = NO
Notice:  AqualinkD: Serial FTDI low latency  = YES
Notice:  AqualinkD: RS485 Frame delay        = 4ms
Notice:  AqualinkD: Config BTN Filter_Pump   = label Filter Pump     | 
Notice:  AqualinkD: Config BTN Spa_Mode      = label Spa Mode        | 
Notice:  AqualinkD: Config BTN Aux_1         = label Cleaner         | 
Notice:  AqualinkD: Config BTN Aux_2         = label Waterfall       | 
Notice:  AqualinkD: Config BTN Aux_3         = label Spa Blower      | 
Notice:  AqualinkD: Config BTN Aux_4         = label Pool Light      | 
Notice:  AqualinkD: Config BTN Aux_5         = label Spa Light       | 
Notice:  AqualinkD: Config BTN Aux_6         = label NONE            | 
Notice:  AqualinkD: Config BTN Aux_7         = label NONE            | 
Notice:  AqualinkD: Config BTN Pool_Heater   = label Pool Heater     | 
Notice:  AqualinkD: Config BTN Spa_Heater    = label Spa Heater      | 
Notice:  AqualinkD: Config BTN Solar_Heater  = label Solar Heater    | 
Notice:  NetService:Starting network services thread
Notice:  RS Serial: Port /dev/ttyUSB0 low latency mode is set
Notice:  NetService:Starting web server on port 80
Notice:  AqualinkD: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Notice:  AqualinkD: Waiting for Control Panel probe
Notice:  NetService:Starting MQTT client to (null)
```

sapozd, 2024-09-08:
Ah after fidgeting with the wires a bit more I finally got the serial logger to display something!

sfeakes, 2024-09-08:
Glad it's working.  The connections on RS485 are usually the problem.

closed 2024-12-08 by sfeakes