aqualinkd/AqualinkD #345: MQTT Processing Last Command Only

twemperor, 2024-09-09:
I'm using OpenHAB over MQTT to send commands and read status from AqualinkD. Overall, I have to say the service is amazing, and incredibly adaptive and configurable to all of the various hardware configurations. I have the iAqualink interface, but understand that my AqualinkD configuration is using the PDA protocol. What I've noticed is that when multiple MQTT commands are sent, only the last is serviced. In the log snippet below, I set SPA Mode ON, Filter Pump ON, and Spa Heater ON, but only the SPA Heater is enabled.

`
Info:  NetService:MQTT: Published id=8133: aqualinkd/Display_Message
Info:  AllButton: Queue send '0x04' to controller (programming)
Info:  NetService:MQTT: Published id=8132: aqualinkd/Spa_Heater/timer/duration 0
Info:  NetService:MQTT: Published id=8131: aqualinkd/Spa_Heater/timer 0
Info:  NetService:MQTT: Published id=8130: aqualinkd/Spa_Heater/enabled 1
Info:  NetService:MQTT: Published id=8129: aqualinkd/Spa_Heater 0
Info:  AllButton: Queue send '0x04' to controller (programming)
Info:  PDA:       PDA Device On/Off, found device 'SPA HEAT', changing state
Info:  AllButton: Queue send '0x05' to controller (programming)
Info:  AllButton: Queue send '0x05' to controller (programming)
Info:  AllButton: Queue send '0x05' to controller (programming)
Info:  AllButton: Queue send '0x05' to controller (programming)
Info:  NetService:MQTT: Published id=8128: aqualinkd/Spa_Mode/timer/duration 0
Info:  NetService:MQTT: Published id=8127: aqualinkd/Spa_Mode/timer 0
Info:  NetService:MQTT: Published id=8126: aqualinkd/Spa_Mode 0
Info:  NetService:MQTT: Published id=8125: aqualinkd/Spa_Mode/delay 0
Info:  NetService:MQTT: Published id=8124: aqualinkd/Filter_Pump/timer/duration 0
Info:  NetService:MQTT: Published id=8123: aqualinkd/Filter_Pump/timer 0
Info:  NetService:MQTT: Published id=8122: aqualinkd/Filter_Pump 0
Info:  NetService:MQTT: Published id=8121: aqualinkd/Filter_Pump/delay 0
Info:  AllButton: Queue send '0x05' to controller (programming)
Info:  PDA:       PDA Device On/Off, device 'SPA HEAT', state 0
Info:  Panl&Prog: Programming: Switch PDA device on/off, 18
Info:  NetService:MQTT: Published id=8120: aqualinkd/Spa_Mode/timer/duration 0
Info:  NetService:MQTT: Published id=8119: aqualinkd/Spa_Mode/timer 0
Info:  NetService:MQTT: Published id=8118: aqualinkd/Spa_Mode 1
Info:  NetService:MQTT: Published id=8117: aqualinkd/Spa_Mode/delay 0
Info:  NetService:MQTT: Published id=8116: aqualinkd/Filter_Pump/timer/duration 0
Info:  NetService:MQTT: Published id=8115: aqualinkd/Filter_Pump/timer 0
Info:  NetService:MQTT: Published id=8114: aqualinkd/Filter_Pump 1
Info:  NetService:MQTT: Published id=8113: aqualinkd/Filter_Pump/delay 1
Info:  AllButton: Queue send '0x04' to controller (programming)
Info:  PDA:       PDA Device On/Off, found device 'FILTER PUMP', changing state
Info:  AllButton: Queue send '0x02' to controller (programming)
Info:  PDA:       PDA Device On/Off, device 'FILTER PUMP', state 0
Info:  Panl&Prog: Programming: Switch PDA device on/off, 18
Error: PDA:       PDA Device On/Off: SPA on - wait for CMD_MSG_LONG
Info:  AllButton: Queue send '0x04' to controller (programming)
Info:  PDA:       PDA Device On/Off, found device 'SPA', changing state
Info:  AllButton: Queue send '0x05' to controller (programming)
Notice:Panl&Prog: Starting programming thread 'Switch PDA device on/off'
Info:  Panl&Prog: received 'ON' for 'Spa_Heater', turning 'ON'
Info:  Panl&Prog: Device request type 'OnOff' for deviceindex 10 'SPA HEAT' of value 1 from 'MQTT'
Notice:Panl&Prog: Starting programming thread 'Switch PDA device on/off'
Info:  Panl&Prog: received 'ON' for 'Filter_Pump', turning 'ON'
Info:  Panl&Prog: Device request type 'OnOff' for deviceindex 0 'FILTER PUMP' of value 1 from 'MQTT'
Info:  NetService:MQTT: Published id=8112: aqualinkd/Display_Message Programming: setting device on/off
Info:  AllButton: Queue send '0x04' to controller (programming)
Info:  AllButton: Queue send '0x06' to controller (programming)
Info:  PDA:       PDA Device On/Off, device 'SPA', state 0
Info:  Panl&Prog: Programming: Switch PDA device on/off, 18
Notice:Panl&Prog: Starting programming thread 'Switch PDA device on/off'
Info:  Panl&Prog: received 'ON' for 'Spa_Mode', turning 'ON'
Info:  Panl&Prog: Device request type 'OnOff' for deviceindex 1 'SPA' of value 1 from 'MQTT'
I`

sfeakes, 2024-09-09:
It looks like you have button 2 set to “Spa Mode” rather than “Spa” and the two are different / do different things. I can see why you have that, looks like the example aqualinkd.conf has that set to spa mode, which would work fine on an RS panel, but not a PDA panel.

Can you try changing that, and if you still have the error, put AqualinkD into debug mode and post the output.

twemperor, 2024-09-13:
I tried SPA Mode instead of SPA, but AqualinkD wouldn't even start since in the PDA controller, button 2 is labelled SPA in my setup. I turned on Debug, and pasted below. I removed the redundant RS Timing messages for readability.

Tracing through the debug messages, it looks to me like maybe it's the order they're being processed? Turning on SPA mode appears to put the filter pump in a disabled state while the valves move. SPA mode turns off (because the filter isn't on?). I re-ordered the commands and was able to get the behavior I want.

I have a similar issue when I send commands to turn on the Pool Light and SPA Light at the same time-only one does. I assume there is a similar issue while the lights are toggling to the right color? I'll continue to debug, but appreciate any suggestions.

Thanks!

Sep 12 21:29:29 Debug:  NetService:MQTT: topic aqualinkd/Spa_Mode/set 1
Sep 12 21:29:29 Debug:  NetService:MQTT: URI Request 'Spa_Mode/set': value 1.00
Sep 12 21:29:29 Info:   Panl&Prog: Device request type 'OnOff' for deviceindex 1 'SPA' of value 1 from 'MQTT'
Sep 12 21:29:29 Info:   Panl&Prog: received 'ON' for 'Spa_Mode', turning 'ON'
Sep 12 21:29:29 Notice: Panl&Prog: Starting programming thread 'Switch PDA device on/off'
Sep 12 21:29:29 Info:   Panl&Prog: Programming: Switch PDA device on/off, 18
Sep 12 21:29:29 Debug:  Panl&Prog: Thread 18,0x419dde70 is active (Switch PDA device on/off)
Sep 12 21:29:29 Info:   PDA:       PDA Device On/Off, device 'SPA', state 0
Sep 12 21:29:29 Debug:  PDA:       PDA Device programmer request for menu 19
Sep 12 21:29:29 Debug:  PDA:       PDA Device programmer looking for menu text 'EQUIPMENT ON/OFF' (limit=16)
Sep 12 21:29:29 Debug:  PDA:       find_pda_menu_item i=4 idx=9 min=4 max=9 boost=0
Sep 12 21:29:29 Debug:  AllButton: Queue empty!
Sep 12 21:29:29 Debug:  AllButton: Queue empty!
Sep 12 21:29:29 Info:   AllButton: Queue send '0x06' to controller (programming)
Sep 12 21:29:29 Debug:  PDA:       waitForPDAMessageHighlight index 9
Sep 12 21:29:29 Debug:  PDA:       waitForPDAMessageHighlight last = 0x02 : index 4 : (1 of 10)
Sep 12 21:29:29 Debug:  NetService:MQTT: topic aqualinkd/Filter_Pump/set 1
Sep 12 21:29:29 Debug:  NetService:MQTT: URI Request 'Filter_Pump/set': value 1.00
Sep 12 21:29:29 Info:   Panl&Prog: Device request type 'OnOff' for deviceindex 0 'FILTER PUMP' of value 1 from 'MQTT'
Sep 12 21:29:29 Info:   Panl&Prog: received 'ON' for 'Filter_Pump', turning 'ON'
Sep 12 21:29:29 Notice: Panl&Prog: Starting programming thread 'Switch PDA device on/off'
Sep 12 21:29:29 Debug:  Panl&Prog: Thread 0x7610d950 (Switch PDA device on/off) sleeping, waiting for thread 0x419dde70 (Switch PDA device on/off) to finish
Sep 12 21:29:29 Debug:  NetService:MQTT: topic aqualinkd/Spa_Heater/set 1
Sep 12 21:29:29 Debug:  NetService:MQTT: URI Request 'Spa_Heater/set': value 1.00
Sep 12 21:29:29 Info:   Panl&Prog: Device request type 'OnOff' for deviceindex 10 'SPA HEAT' of value 1 from 'MQTT'
Sep 12 21:29:29 Info:   Panl&Prog: received 'ON' for 'Spa_Heater', turning 'ON'
Sep 12 21:29:29 Notice: Panl&Prog: Starting programming thread 'Switch PDA device on/off'
Sep 12 21:29:29 Debug:  Panl&Prog: Thread 0x761ac4c8 (Switch PDA device on/off) sleeping, waiting for thread 0x419dde70 (Switch PDA device on/off) to finish
Sep 12 21:29:29 Debug:  RS Timings:Time between packets (1726190976.000 sec)
Sep 12 21:29:29 Debug:  PDA:       Kicking PDA thread 18,0x419dde70
Sep 12 21:29:29 Debug:  PDA:       waitForPDAMessageHighlight last = 0x02 : index 4 : (2 of 10)
Sep 12 21:29:29 Info:   NetService:MQTT: Published id=18153: aqualinkd/Display_Message Programming: setting device on/off
Sep 12 21:29:29 Debug:  NetService:MQTT: Ignore aqualinkd/Display_Message Programming: setting device on/off
Sep 12 21:29:29 Debug:  RS Timings:Time between packets (1726190976.000 sec)
Sep 12 21:29:29 Info:   PDA:       PDA Menu Line 0 =      THU  9:29PM
Sep 12 21:29:29 Info:   PDA:       PDA Menu Line 1 = AIR
Sep 12 21:29:29 Info:   PDA:       PDA Menu Line 2 =  74`
Sep 12 21:29:29 Info:   PDA:       PDA Menu Line 3 =
Sep 12 21:29:29 Info:   PDA:       PDA Menu Line 4 = POOL MODE    OFF
Sep 12 21:29:29 Info:   PDA:       PDA Menu Line 5 = POOL HEATER  OFF
Sep 12 21:29:29 Info:   PDA:       PDA Menu Line 6 = SPA MODE     OFF
Sep 12 21:29:29 Info:   PDA:       PDA Menu Line 7 = SPA HEATER   OFF
Sep 12 21:29:29 Info:   PDA:       PDA Menu Line 8 = MENU
Sep 12 21:29:29 Info:   PDA:       PDA Menu Line 9 = EQUIPMENT ON/OFF
Sep 12 21:29:29 Debug:  PDA:       PDA Menu highlighted line#=9 line='EQUIPMENT ON/OFF'
Sep 12 21:29:29 Debug:  PDA:       Kicking PDA thread 18,0x419dde70
Sep 12 21:29:29 Debug:  PDA:       waitForPDAMessageHighlight last = 0x08 : index 9 : (3 of 10)
Sep 12 21:29:29 Debug:  PDA:       waitForPDAMessageHighlight: received index '9'
Sep 12 21:29:29 Debug:  AllButton: Queue empty!
Sep 12 21:29:29 Info:   AllButton: Queue send '0x04' to controller (programming)
Sep 12 21:29:29 Debug:  PDA:       PDA Device programmer selected menu item 'EQUIPMENT ON/OFF'
Sep 12 21:29:29 Debug:  PDA:       waitForPDAMessageType  0x09
Sep 12 21:29:29 Debug:  PDA:       waitForPDAMessageType 0x09, last message type was 0x08 (1 of 10)
Sep 12 21:29:29 Debug:  NetService:MQTT: Message publishing acknowledged (msg_id: 18153)
Sep 12 21:29:29 Debug:  RS Timings:Time between packets (1726190976.000 sec)
Sep 12 21:29:30 Debug:  PDA:       Kicking PDA thread 18,0x419dde70
[... cut ...]
Sep 12 21:29:38 Debug:  JandyDvce: ControlPanel request Pump ID 0x7a get watts
Sep 12 21:29:38 Debug:  JandyDvce: Last panel info RPM:0 GPM:0 WATTS:0
Sep 12 21:29:38 Debug:  RS Timings:Time between packets (1726190976.000 sec)
Sep 12 21:29:39 Info:   NetService:MQTT: Published id=18174: aqualinkd/Display_Message
Sep 12 21:29:39 Debug:  NetService:MQTT: Ignore aqualinkd/Display_Message
Sep 12 21:29:39 Debug:  NetService:MQTT: Message publishing acknowledged (msg_id: 18174)
Sep 12 21:29:39 Debug:  RS Timings:Time between packets (1726190976.000 sec)`

sfeakes, 2024-09-13:
Looks like you sent commands in the following order "spa" "filter pump" "heater".
AqualinkD did all those commands. (It simply follows the PDA menu, as that's all it can do on a PDA mode).
So that means.
```
Received Spa on (checked spa is off and start command sequence to turn it on)
Received Filter Pump on (checked filter pump is off, and queue a sequence of commands)
Received Heater on (checked it's off, and queue a sequence of commands)
Finished running through Spa commands (turned spa on) This will also turn Filter Pump on.
Ran through Filter Pump, since this was now on, it turned it off.
Ran through Spa Heater and turned it on.
```

I would recommend sending "Filter Pump" first in this case. OR better yet simply send "spa" & "spa heater".

Spa vs Spa Mode are different on many panels (depending on version). Spa mode is more of a "virtual button", where it's not assigned to a specific circuit on the panel, but rather 2 commands. As it turns on Spa & Spa Heater.

Colorlight are not supported in PDA mode.  So that's probable where your problem lies on turning the light on.

closed 2024-12-08 by sfeakes