aqualinkd/AqualinkD #351: Aqualinkd not reflecting controller status

sts098, 2024-09-22:
I am running a pda-ps8 combo with rev 7.1.0.   Running current development version 2.4.1 dev 0.1.  My system has the pda-ps8, Jandy 18 channel RF receiver, IAquaLink, and a Jandy variable speed pump. 

Operating well, not seeing checksum warnings in Aqualinkd manager.

Operating well, until schedule in pda turns off pump.  At that point, Aqualinkd/manager still shows pump on (bright blue).  Speed shows last speed, watts drops from 980 or so, and shows 4 (likely watts to keep board energized).  

Similarly, turning pump off through IAquaLink while cleaner is running (message in IAquaLink says pump will remain on while cleaner is running).  Aqualinkd shows cleaner on delay (filter is actually on delay). Turning off cleaner in IAquaLink, shuts down cleaner, then filter pump.  Aqualinkd still shows cleaner status delay, pump on speed 3450 (cleaning speed) again, watts on 4, pump on. 

Seems Aqualinkd pump off status went sent from pda controller and not understanding delayed device.

sfeakes, 2024-09-22:
Can you post your` aqualinkd.conf` file.  A lot will depend on settings you are using and I'd like to replicate as much as possible.

sts098, 2024-09-23:
Attached is my conf file.  Blanked out MQQT Stuff.  Otherwise, this is the working version.  Thanks for looking at it.  
[aqualinkd.txt](https://github.com/user-attachments/files/17098706/aqualinkd.txt)

sfeakes, 2024-09-23:
Thanks for that, I think I know where the problem lies.  When iAqualink2 device is attached it changes the order of messages on certain panel revisions, normally this is not an issue on RS panel since I don’t use the messages for status, but in PDA mode that’s the only option available.  I can’t replicate this exactly, so I’lol need a little more information from you if you don’t mind.
Can you.

- use aqmanager and under Debug Masks, set PDA to on.
- Run through commands that highlight the problem.
- Download log file and post it (probably want at least 1k lines)

You may have to do the above twice for both issues.

When 4.2.1 get’s released, I hope to have a better PDA option for this as I can sniff & read what’s being sent to iAqualink2 device and decode that since that protocol has a better way of defining when something is on/off that reading text messages. I just hope Jandy did not limit the iAqualink2 protocol on PDA only panels.

sts098, 2024-09-23:
I ran the testing twice.  Attached are log files for pump and cleaner tests.  I will send screen shots separately.  

[Pump Off 2.log](https://github.com/user-attachments/files/17102639/Pump.Off.2.log)
[Pump Off 1.log](https://github.com/user-attachments/files/17102640/Pump.Off.1.log)
[Cleaner test 2.log](https://github.com/user-attachments/files/17102641/Cleaner.test.2.log)
[Cleaner test 1.log](https://github.com/user-attachments/files/17102642/Cleaner.test.1.log)

sts098, 2024-09-23:
Here are screenshots referenced above.

[image]
[image]
[image]
[image]

sfeakes, 2024-09-26:
@sts098   Thanks for the information.  The bug is fixed and will be posted in 2.4.1 (Dev 0.2) in a few days, and released with 2.4.1

sts098, 2024-10-10:
I have been running 2.4.1 dev 0.2 since its release.  It has corrected the issue above.  And the system performs reliably.

closed 2024-12-08 by sfeakes