aqualinkd/AqualinkD #353: Increased checksum errors with v2.4.1 dev 0.2

sts098, 2024-10-10:
Aqualinkd is running well.  With the Aqualinkd Management Console,  I have noticed an increase in checksum errors and bad packets.  It does not seem to interfere with the operation.  

Under v2.4.1 dev 0.1 Jandy checksum errors were very infrequent, going 10’s of minutes even hours without a single error not even filling the console window after a day or more.  With dev 0.2 these became much more frequent.  I think of checksum errors as communication issues.  I am running with a PDA- PS8 with ver 7.1.0 and an ew11 rather than hard wired serial port, but this was the same set up with dev 0.1.   

Any idea what may be causing the increase?

should I be concerned about it?

sfeakes, 2024-10-10:
I am sure this is EW11 and your wifi.   I’ve done a lot of playing with EW11 on my own setup and have a lot of logs from someone else with EW11.  There are some socat command changes I posted in the EA11 thread that will help. But ultimately you’ll need to start looking at wifi.  I do think moving EW11 and socat from TCP to UDP might help but I’ve not managed to get that to work

sts098, 2024-10-10:
Ok, thanks for the quick response.

sts098, 2024-10-23:
I just switched to version 2.5.0 dev 0.1 and the checksum errors are gone.  Still using the ew11.  No changes to the system except — read_RS485_iAqualink = yes.

closed 2024-12-08 by sfeakes