aqualinkd/AqualinkD #359: Home Assistant Discovery Messages Are Not Retained

shbatm, 2024-11-08:
Just tried upgrading from several versions back and in the process I wanted to switch to using the Home Assistant Discovery messages instead of the manual config. I noticed though that the discovery messages are not set to retain, which causes them to get lost on the Home Assistant side if Home Assistant is restarted but the broker and Aqualinkd are not (e.g. Aqualink maintains broker connection while HA loses it).

See recommendations: https://www.home-assistant.io/integrations/mqtt/#discovery-messages-en-availability

I was in the process of doing a PR to tweak a couple of the discovery messages (e.g. add the current action to the climate entities) which is straighforward enough, but it looks like MQTT uses a global flag for retention so I wasn't sure what the best path forward would be.

sfeakes, 2024-11-08:
This is a very good point that I overlooked. I’ll fix it in next update.

shbatm, 2024-11-08:
No you're right, it's an `0x1` flag for the constant. I don't know why but MQTT-Explorer wasn't showing the topics as retained when I was testing yesterday, reconnecting today and they are. Closing this.

closed 2024-11-08 by shbatm

sfeakes, 2024-11-08:
When you reboot HA did you also reboot MQTT broker as well?  I noticed that the default HA MQTT broker has retain set to memory and not disk. So that might explain it.   For many reasons I don’t like the default HA implementation of Mosquito, so I run Mosquito outside HA.

shbatm, 2024-11-09:
It was just a HA reboot, but I was watching the message stream in a separate app, I think it was some combo of me messing with too many things at once (testing changes, removing old manual config, misreading the code, etc), seems to be working fine now.

Also just a comment on the HA Broker: it flushes to disk periodically (`Saving in-memory database to /data//mosquitto.db.`) but I'm almost positive the reason they did it that way was reducing SD card writes and corrupting cards--but I agree, it risks dropping data  if the broker crashes.