aqualinkd/AqualinkD #361: Feature Request: Home Assistant MQTT support adding MAC Address to device info

shbatm, 2024-11-16:
Would you be supportive of adding a config item to be able to specify the MAC Address for use with MQTT HASSIO Device Discovery?

Something like `mqtt_device_mac_address=aa:bb:cc:dd:ee:ff`

This would allow the "[connections](https://www.home-assistant.io/integrations/binary_sensor.mqtt/#connections)" property of the device to be sent, and on the Home Assistant side, this enables router device trackers to be linked together with the AqualinkD device and show if it's connected or not. 

I'm able to implement here, but wanted to see if it would be supported first.

sfeakes, 2024-11-17:
Yes I'd be supportive.  I don't use home assistant, so not exactly sure what benefit this provides.
Please use the function `bool mac(char *buf, int len)` in config.c to obtain the MAC address. (just make it public by adding it to config.h
This will simply return the first valid MAC it finds, at the moment it's only used for a unique ID for MQTT, so if it get's the wrong MAC it's not an issue.  For this case though, the wrong MAC would be an issue, so we will need to look into obtaining the correct MAC if a system has multiple NIC's.

sfeakes, 2024-11-17:
Update.  I just realized the Mac function would not return the result you wanted, as it didn't use a `:` for the delimiter.
I modified the function, and also since I was there, made the changes as I decided a config option would be best for this at the moment.

Please pull latest source `3.5.1 (dev)` and use `mqtt_hass_discover_use_mac = yes` in aqualinkd.conf.    See if that gives you what you want, if not please modify as appropriate.

closed 2024-11-17 by sfeakes

shbatm, 2024-11-17:
Thanks! The code worked but was grabbing the wrong MAC address (it was grabbing the `eth0` instead of `wlan0`). I pushed #362 which checks the flags to get the first interface that actually has a connection and uses that MAC instead.

I'm not sure the config flag is actually needed -- it will not hurt anything on the HA side if it has the connection all the time in the discovery payload, it will only do anything on the HA side if there are multiple devices sharing the MAC address.

Thanks for supporting. Just for examples: this enables automatic combining of devices in Home Assistant--I have NodeRED running on the same Rock Pi S with a 2nd RS485 adapter to control the Century Link VS Pump which isn't connected to the Aqualink control system (retrofit and RS panel is too old to support)--this lets AqualinkD, NodeRED, and the network device all show up as a single "Device" in Home Assistant:

[image]
[image]

sfeakes, 2024-11-17:
That looks great.  The reason I made it config is if on a system with multiple NIC's and it pic's the wrong NIC I was thinking it may mess-up HA.  So as a config option people can turn it off if needed.   I haven't tested when happens if it posts the incorrect MAC since this is dev at moment. I may remove the config option if it's not needed.

shbatm, 2024-11-17:
On the HA side, the only thing that should happen with an incorrect MAC is you would have multiple Devices instead of being combined. "Devices" as far as HA is concerned is just a bunch of individual entities that can be grouped by a common `identifier` and/or `connection` tuple -- even a junk mac address with just get spit back out: https://github.com/home-assistant/core/blob/43235713c7010c8f3676e53a780b792ee4fbad36/homeassistant/helpers/device_registry.py#L462

shbatm, 2024-11-18:
Tacking on some additional screen shots just to share back what this has helped me build. Thanks again for the help!

[image]

[image]