aqualinkd/AqualinkD #367: JandyDvce: Jandy Pump found at ID 0x78 with RPM 2900, but not configured, information ignored!

heffneil, 2024-12-05:
I have added this:

# Pump configuration
vsp_device_id = 0x78

to my config file but it continues to say that I don't have it configured. Any ideas?

sfeakes, 2024-12-05:
You can ignore the message it’s only a warning that you’ll get slower pump inflation.
But to get rid of the message you need to assign the pump to an appropriate button with.
(You may need to change 01)
button_01_pumpID=0x78

you should also assign the pump type as well.  Information is in the wiki.

heffneil, 2024-12-05:
I think I Have an LX heater by the way.  If you explain what and how you need it I can try to help?

sfeakes, 2024-12-08:
@heffneil Can you please post the output to serial logger.  The easiest way to run serial logger is to go to aqmanger, expand the "Serial Logger" menu on the left and hit run.

That should show is if the heater is connected to RS485 bus and what the ID is, from their their may be some more commands to run, but I'll need the serial logger output first.

heffneil, 2024-12-08:
Ok I am turning down my logging I had debugging on and restarting.  When I restart I do see this:

``

Error: AqualinkD: Unknown config parameter 'vsp_type = jandy'
Error: AqualinkD: Unknown config parameter 'vsp_device_id = 0x78'
Error: AqualinkD: Unknown config parameter 'heater_setpoint_min = 60'
Error: AqualinkD: Unknown config parameter 'heater_setpoint_max = 104'
Info:  Error:   AqualinkD: Unknown config parameter 'vsp_type = jandy'
Info:  Error:   AqualinkD: Unknown config parameter 'vsp_device_id = 0x78'
Info:  Error:   AqualinkD: Unknown config parameter 'heater_setpoint_min = 60'
Info:  Error:   AqualinkD: Unknown config parameter 'heater_setpoint_max = 104'
Info:  Error:   AqualinkD: Unknown config parameter 'heater_device_id = 0x69'
Info:  Error:   AqualinkD: Unknown config parameter 'swg_device_id = 0x50'
Error: AqualinkD: Unknown config parameter 'heater_device_id = 0x69'
Error: AqualinkD: Unknown config parameter 'swg_device_id = 0x50'
Notice:AqualinkD: Exit!
Debug: RS Timings:Time between packets (0.109 sec)
``

sfeakes, 2024-12-08:
None of those lines are valid for aqualinkd.conf.  That's why you are getting the error.
ie you have a line in aqualinkd.conf that has `vsp_type = jandy` and that's not a valid configuration option.

The below has all the valid options.
https://github.com/sfeakes/AqualinkD/wiki#Configuration

Below is my best GUESS at what you might be 
```
button_01_pumpType = JANDY ePUMP
button_01_pumpID = 0x78
button_01_pumpIndex = 1    <-- This is a total guess, check your control panel
```
None of the other options are even remotely valid.
AqualinkD will assume only one SWG and Heater, so it doesn't need to know the ID, it can calculate that. You can have multiple VSP pumps, hence AqualinkD needs to know what button to assign the pump to.
Heater max/min set points are fixed at the control panel, so again you can't change them

heffneil, 2024-12-08:
thats what you get when you rely on ChatGPT :)

closed 2025-01-29 by sfeakes