aqualinkd/AqualinkD #372: Spa Mode not functioning correctly in 2.5.1

shbatm, 2024-12-23:
Having trouble with Spa Mode in the latest version, looks like it may have been related to some cleanup from 2.5.0 and `spa` vs `spa_mode` getting mixed and matched.

On startup, spa mode is off and it shows up correctly in Home Assistant via MQTT and in the Web Interface I have a "Spa Mode" button. The MQTT discovery is pushing `Spa_Mode` topic (and this is still referenced in the old manual HASSIO config).

I am able to successfully change to Spa Mode using the web interface; however, as soon as it switches over, the Spa Mode button dissapears from the interface.

On MQTT, it is only reporting the state to the `Spa` topic, not `Spa_Mode`.

Startup / Panel Log:

```shell
Dec 23 08:54:06 aqualink-rps aqualinkd[2094]: AqualinkD: Aqualink Daemon v2.5.1 (dev)
Dec 23 08:54:06 aqualink-rps aqualinkd[2094]: AqualinkD: Panel set to RS-6 Combo Pool/Spa
Dec 23 08:54:06 aqualink-rps aqualinkd[2094]: AqualinkD: Config log_level         = 6
Dec 23 08:54:06 aqualink-rps aqualinkd[2094]: AqualinkD: Config device_id         = 0x0a
Dec 23 08:54:06 aqualink-rps aqualinkd[2094]: AqualinkD: Config rssa_device_id    = 0x00
Dec 23 08:54:06 aqualink-rps aqualinkd[2094]: AqualinkD: Config extra_device_id   = 0x00
Dec 23 08:54:06 aqualink-rps aqualinkd[2094]: AqualinkD: Config BTN Filter_Pump   = label Filter Pump     |
Dec 23 08:54:06 aqualink-rps aqualinkd[2094]: AqualinkD: Config BTN Spa           = label Spa Mode        |
Dec 23 08:54:06 aqualink-rps aqualinkd[2094]: AqualinkD: Config BTN Aux_1         = label Cleaner         |
Dec 23 08:54:06 aqualink-rps aqualinkd[2094]: AqualinkD: Config BTN Aux_2         = label Spa Blower      |
Dec 23 08:54:06 aqualink-rps aqualinkd[2094]: AqualinkD: Config BTN Aux_3         = label Pool Light      |
Dec 23 08:54:06 aqualink-rps aqualinkd[2094]: AqualinkD: Config BTN Aux_4         = label Spa Light       |
Dec 23 08:54:06 aqualink-rps aqualinkd[2094]: AqualinkD: Config BTN Aux_5         = label Waterfall       |
Dec 23 08:54:06 aqualink-rps aqualinkd[2094]: AqualinkD: Config BTN Pool_Heater   = label Pool Heater     |
Dec 23 08:54:06 aqualink-rps aqualinkd[2094]: AqualinkD: Config BTN Spa_Heater    = label Spa Heater      |
Dec 23 08:54:06 aqualink-rps aqualinkd[2094]: AqualinkD: Config BTN Extra_Aux     = label Fill Line       |
Dec 23 08:54:23 aqualink-rps aqualinkd[2095]: AllButton: Control Panel version 6521 REV H
Dec 23 08:54:23 aqualink-rps aqualinkd[2095]: AllButton: Control Panel revision H
```

sfeakes, 2024-12-23:
Thanks for the PR. Spa and Spa_Mode used to be the same depending on panel version, but now with virtual buttons they are two separate things.
BTW, you don;t have to manually configure HA anymore, if you set `mqtt_hassio_discover_topic = homeassistant` in your aqualinkd.conf, AqualinkD will configure HA for you by pushing discoverable devices.  This way you shouldn't have issues in the future.

closed 2024-12-23 by sfeakes

shbatm, 2024-12-23:
> BTW, you don;t have to manually configure HA anymore, if you set `mqtt_hassio_discover_topic = homeassistant` in your aqualinkd.conf, AqualinkD will configure HA for you by pushing discoverable devices. This way you shouldn't have issues in the future.

I'm using the Autodiscovery, just cleaning up the old file since I saw it while troubleshooting. My issue with the auto-discovery was a stale retained config entry for `Spa_Mode` on the MQTT Broker, in addition to the new one created at `Spa`--my Home Assistant dashboard was using the old one and it wasn't getting updated when the mode changed.

PR #373 (2024-12-23): Update HASSIO.Implementation.txt for Spa Mode
Change Spa_Mode to Spa for state topic. Fixes #372.