aqualinkd/AqualinkD #374: Setting blower timer by default?

42Cyborg, 2024-12-29:
Hello, I don't know if this should be marked as an issue as it may already be possible and I am missing it. If not, it is really more of a suggestion, but I would like to be able to have a timer started by default when I turn on a specific item (blower) from the web interface. I always long-press and set a timer when I use it, but my family tends to just turn it on and it runs for hours until I realize.
Is there a way to have it automatically only run for an hour when turned on from the web interface? I am on the latest version. 
Thanks!

aqualinkd, 2024-12-30:
AqualinkD doesn't support a specific (turn off after X time), unless you set the times as you mentioned.
The Jandy control panel also has an option called "Device Runtime" that should do what you want. 
If you have AqualinkD integrated into Homeassistant, Domoticz, HomeKit, or any other home automation hub, then each of those also support that kind of functionality you are after.

42Cyborg, 2024-12-30:
Thank you so much, that works perfectly!

closed 2024-12-30 by 42Cyborg