aqualinkd/AqualinkD #376: Iaqualink and SWG issue and question

doomngloom, 2025-01-06:
Hi. I just recently setup the aqualinkd and am lovng how it is working so far. I am coming up with what i hope are 2 simple issue as well as a question.
Issue 1: when aqualinkd is running then the iaqualink is sometimes online and sometimes offline and cannot be used. When its online then sometimes i can login to iaqualink but then it errors out with a Connection error Connection Lost and wont work while other time it works just fine. 
The items i have enabled in the config for iaqualink are below: the last one read_RS485_iAqualink i tried with it commented out, uncommented with yes and uncommented with no. none of them make a difference.

extended_device_id=0x30
enable_iaqualink=yes
read_RS485_iAqualink = no

Below is the data from the serial logger. I have tried changing the extended device id to macth what is shown in case that was the issue. when i change it then the service wont start. says there is an error in the config.

Issue 2: I am getting constant warnings for the SWG. Im not sure what they are or why they are there. the SWG seems to be working fine but seeing warnings bothers me. id love to get rid of it if possible by resolving whatever that issue is. Below is the debug logging for before and after one of the warnings.

Info:  iAQ Touch: Page: Status (diff ID) | 0x2a
Info:  iAQ Touch: Not sure where to add Button 1  - LoadingPage = Status (diff ID)
Info:  iAQ Touch: Page: Devices #2 | 0x35
Info:  AllButton: RS Message :- 'AquaPure 35%'
Info:  iAQ Touch: Page: Devices #1 | 0x36
Info:  AllButton: RS Message :- 'Pool Temp  73 F'
Info:  JandyDvce: Jandy Pump Status message = WATTS 369
Info:  JandyDvce: ControlPanel request Pump ID 0x78 get watts
Info:  iAQ Touch: Page: Devices #1 | 0x36
Info:  AllButton: RS Message :- 'Air Temp 84 F'
Info:  iAQ Touch: Home Status page 05| Air Temp
Info:  iAQ Touch: Home Status page 04| Pool Temp
Info:  iAQ Touch: Home Status page 01| 84
Info:  iAQ Touch: Home Status page 00| 73
Info:  iAQ Touch: Page: HOME | 0x01
Info:  iAQ Touch: Home Status page 05| Air Temp
Info:  iAQ Touch: Home Status page 04| Pool Temp
Info:  iAQ Touch: Home Status page 01| 84
Info:  iAQ Touch: Home Status page 00| 73
Info:  iAQ Touch: Page: HOME | 0x01
Info:  iAQ Touch: Status page 05| Salt 3000 PPM
Info:  iAQ Touch: Status page 04| AquaPure 35%
Info:  iAQ Touch: Status page 03|   Watts: 369
Info:  iAQ Touch: Status page 02|     RPM: 2000
Info:  iAQ Touch: Status page 01| Jandy ePUMP   1
Info:  iAQ Touch: Status page 00| Filter Pump
Info:  iAQ Touch: Page: Status (diff ID) | 0x2a
Info:  iAQ Touch: Not sure where to add Button 1  - LoadingPage = Status (diff ID)
Info:  iAQ Touch: Page: Devices #2 | 0x35
Warning:JandyDvce: Ignoring set SWG device to state '0x60', state is unknown
Info:  AllButton: RS Message :- '2:07 PM'
Info:  iAqualink2:Queue cmd, size 2,  queu length=3
Info:  iAqualink2:Queue cmd, size 2,  queu length=2
Info:  iAqualink2:Queue cmd, size 2,  queu length=1
Debug: NetService:WS: URI Request 'setloglevel': value 6.00
Debug: NetService:WS: Message - Key 'value' Value '6'
Debug: NetService:WS: Message - Key 'uri' Value 'setloglevel'
Debug: RS Timings:Time between packets (0.014 sec)
Debug: RS Timings:Time between packets (0.056 sec)
Debug: RS Timings:Time between packets (0.009 sec)
Debug: RS Timings:Time between packets (0.017 sec)
Debug: RS Timings:Time between packets (0.023 sec)
Debug: RS Timings:Time between packets (0.027 sec)

Finally for the question, i have an mqqt item that is being sent from aqualinkd to my mqtt server called aqualinkd/SWG/PPM_f. what is that. i thought at first it was the available salt in my pool which kind of matched a salt test i did (mine showed at 1800 not 1600) but it hasent updated to anything new when i added salt to the pool. additionally at that number of 1649 i expect the system to give an error of low salt as it usually does. at 1800 it doesnt always do that

Thank you. Let me know if you need any additional information

doomngloom, 2025-01-06:
Sorry here are some of the warnings for issue 2 without the debug.

Warning:JandyDvce: Ignoring set SWG device to state '0x70', state is unknown
Warning:JandyDvce: Ignoring set SWG device to state '0x82', state is unknown
Warning:JandyDvce: Ignoring set SWG device to state '0x90', state is unknown
Warning:JandyDvce: Ignoring set SWG device to state '0x50', state is unknown
Warning:JandyDvce: Ignoring set SWG device to state '0x82', state is unknown
Warning:JandyDvce: Ignoring set SWG device to state '0x30', state is unknown
Warning:JandyDvce: Ignoring set SWG device to state '0x82', state is unknown
Warning:JandyDvce: Ignoring set SWG device to state '0x32', state is unknown
Warning:JandyDvce: Ignoring set SWG device to state '0x82', state is unknown
Warning:JandyDvce: Ignoring set SWG device to state '0xc0', state is unknown
Warning:JandyDvce: Ignoring set SWG device to state '0x92', state is unknown
Warning:JandyDvce: Ignoring set SWG device to state '0x62', state is unknown
Warning:JandyDvce: Ignoring set SWG device to state '0x50', state is unknown
Warning:JandyDvce: Ignoring set SWG device to state '0x82', state is unknown

sfeakes, 2025-01-06:
On issue #1.   One one iAqualink device can be enabled and writing. So if you have iAqualink then :

`enable_iaqualink=yes` must be commented out or set to no. -> This is your main problem.
`read_RS485_iAqualink = yes` AqualinkD can still read iAqualink messages, it just can’t write them. So leave this to yes.
`extended_device_id=0x30`. As long as this device ID is free, you can leave it.  (Check withy serial logger).

On issue #2.

Your SWG is returning a state to the control panel that has not been decoded.  You can ignore these messages, it’s more me forgetting to set these to Notice, as they shouldn’t be warnings. (I will update next release).  But can you give me some more detail on exactly what SWG you are using and its version (if you know)?

PPM_f (along with other _f MQTT messages like SWG_f) are simply a deg C to F conversion of the actual number.  Ie if actual PPM was 2300 then aqualinkd would also post PPM_f of 4172.  This is for some automation hubs like homekit. They expect number to be posted in deg C and will automatically do a conversion. (Even though PPM is not a temperature).  So you can simply ignore that if you don’t need it.

doomngloom, 2025-01-06:
Thank you for the quick response. For issue 1 if i set enable_iaqualink=no or comment it out wont i lose the ability to use virtual buttons? it hints at that in the notes. As you said i made the change and iaqualink is working now. In the next version maybe update the notes in the config to state something similiar? i'm going to update mine for myself.
for the SWG. I'm using the Jandy Pro Series TrueClear. Not sure about the version though. some of the labels have been bleached an i dont have the data recorded anywhere,
I was hoping the PPM_f was some secret way Jandy wasnt publishing to see the actual salt numbers since the True Clear doesnt actually give any actual numbers for the available salt it always shows 3000. I'm gonna remove it from my automation since its nothing i can use.

Thank you

sfeakes, 2025-01-06:
I just realized the documentation for `enable_iaqualink=yes` is not complete, as I only added that in the last release, and doesn’t state that you can’t use it if you have an iAqualink device attached.  Sorry about the confusion I will update docs.  It will also mean that you can’t use virtual buttons, but I will fix that in next release.

doomngloom, 2025-01-06:
Thank you. Guess i'll wait for the next version to test that then. I'm satisfied my questions and issues are answered and resolved. you can mark this closed if you want.

I greatly appreciate the help and the program

sfeakes, 2025-01-06:
I just looked over the code, and I did implement that you can you virtual buttons with `enable_iaqualink=no` as long as you have extended_device_id set to any Iaqualink touch ID, So `extended_device_id=0x30` will be fine with virtual buttons.  It will just be a little slower than if you have `enable_iaqualink=yes`

I will update the documentation.  Thanks for noticing that.