aqualinkd/AqualinkD #387: AqualinkD service up but not responding, MQTT up HBv1.9 installed, latest Aqualink installed

woodgieboodgie, 2025-02-21:
[21/02/2025, 11:43:55] [AqualinkD] Loading platform accessory [Pool Light]
[21/02/2025, 11:43:55] [AqualinkD] Loading platform accessory [Solar Pump]
[21/02/2025, 11:43:55] [AqualinkD] Loading platform accessory [Heater]
[21/02/2025, 11:43:55] [AqualinkD] Loading platform accessory [Pool Heater]
[21/02/2025, 11:43:55] [AqualinkD] Loading platform accessory [Spa Heater]
[21/02/2025, 11:43:55] [AqualinkD] Loading platform accessory [Solar Heater]
[21/02/2025, 11:43:55] [AqualinkD] Loading platform accessory [Freeze Protection]
[21/02/2025, 11:43:55] [AqualinkD] Loading platform accessory [Pool Air Temperature]
[21/02/2025, 11:43:55] [AqualinkD] Loading platform accessory [Pool Water Temperature]
[21/02/2025, 11:43:55] [AqualinkD] Loading platform accessory [Spa Water Temperature]
[21/02/2025, 11:43:55] [AqualinkD] Loading platform accessory [Salt Water Generator]
[21/02/2025, 11:43:55] [AqualinkD] Loading platform accessory [Salt Water Generator Percent]
[21/02/2025, 11:43:55] [AqualinkD] Loading platform accessory [SWG Boost]
[21/02/2025, 11:43:55] [AqualinkD] Loading platform accessory [Salt Level PPM]
[21/02/2025, 11:43:55] [AqualinkD] Loading platform accessory [Water Chemistry pH]
[21/02/2025, 11:43:55] [AqualinkD] Loading platform accessory [Water Chemistry ORP]
[21/02/2025, 11:43:55] [AqualinkD] Connecting MQTT broker to 10.0.1.15:1883 with topic aqualinkd
[21/02/2025, 11:43:56] [AqualinkD] Error connecting to AqualinkD service : `request to http://10.0.1.15/api/homebridge failed, reason: connect ECONNREFUSED 10.0.1.15:80`
[21/02/2025, 11:43:56] [AqualinkD] Successfully connected to MQTT broker.
[21/02/2025, 11:43:57] [AqualinkD] AqualinkD is online
[21/02/2025, 11:44:55] [AqualinkD] Error connecting to AqualinkD service : `request to http://10.0.1.15/api/homebridge failed, reason: connect ECONNREFUSED 10.0.1.15:80`

sfeakes, 2025-02-21:
Check your USB adapter and wiring.  AqualinkD is not seeing any messages on the RS485 bus.

woodgieboodgie, 2025-02-21:
Shall do, but just to clarify its by design that AqualinkD shuts down if it cant talk to the usb?

sfeakes, 2025-02-21:
Yes it’s by design.  On startup, if it doesn’t get anything from RS bus, it will shutdown again.  If it had already made a connection to the panel, then looses it, then it will keep trying indefinitely on RS485 errors, but not on all USB errors.

closed 2025-04-14 by sfeakes