aqualinkd/AqualinkD #399: HA Spa Pump Entity Doesn't Work

bcre3306, 2025-04-08:
Hi

The MQTT call for spa pump from HA doesn't seem to be working.  I have the aqualinkd Spa entity and can trigger the switch but it's not making the correct call to MQTT.  I can see in MQTT explorer that its not changing the value from Spa from 0 to 1 but if I create a button in HA to make the call manually via the mqtt service it works fine.  Just wasn't sure if this was something specific with my configuration or if this was a bug.

Thanks again for all the great work!
Brian

sfeakes, 2025-04-08:
It’s easy to have Spa and Spa Mode mixed up.  Check aqualinkd.conf and make sure button_02 is “Spa” and not “Spa Mode”.

bcre3306, 2025-04-08:
Hi Shawn

I believe I have that set correctly.  It works fine in the web gui for AqualinkD, it’s just not working in HA.

sfeakes, 2025-04-08:
The web UI doesn’t use MQTT, and has a direct link to the actual button through a websocket.  MQTT needs to use the button ID. So the case you described web UI works / HA doesn’t can be due to the label.  Can you post your aqualinkd.conf?  I can test using that.

bcre3306, 2025-04-08:
Hi

See attached.

Thank you
Brian

sfeakes, 2025-04-08:
There is no attachment.  You will probably need to use GitHub forum and not reply to email to attach.

bcre3306, 2025-04-08:
Sorry about that, lets try this again. Not sure why but had to change the extension to .txt to attach it.

[aqualinkd.txt](https://github.com/user-attachments/files/19658227/aqualinkd.txt)

sfeakes, 2025-04-09:
I see you are not using AqualinkD’s HA integration.  You must be using your own with a HA configuration.yaml or equivalent.  It’s that yaml that will define what MQTT topics it uses for the spa.
I’d recommend using AqualinkD’s HA integration, information is in the wiki.   

If you still want to use the configuration.yaml, then look in that file for something like the below
Check lines state_topic & command_topic
```
- platform: mqtt
    unique_id: aqualink_spa
    name: "Spa"
    state_topic: "aqualinkd/Spa"
    command_topic: "aqualinkd/Spa/set"
    json_attributes_topic: "aqualinkd/Spa/delay"
    json_attributes_template: "{{ {'delay': value|int} | tojson }}"
    qos: 1
    payload_on: "1"
    payload_off: "0"
    retain: false
    icon: mdi:hot-tub
  

```

bcre3306, 2025-04-09:
Ok thank you for pointing me in the right direction.  The weird thing was it was picking up a bunch of the devices in HA just by pointing it to the MQTT environment integrated into HA.  When I added the hassio line to aqualinkd.conf it brought in the rest of them.  I’ll work on the HA configs today.
Thanks again!!

bcre3306, 2025-04-09:
> If you didn’t have that line in AqualinkD.conf then the entities you had in HA were not created by AqualinkD. Probably created manually by a HA configuration yaml. So you probably have duplicates now. You should delete the old ones to avoid confusion in the future.
>
>
> If you didn’t have that line in AqualinkD.conf then the entities you had in HA were not created by AqualinkD. Probably created manually by a HA configuration yaml. So you probably have duplicates now. You should delete the old ones to avoid confusion in the future.

It only created a few duplicates.  The original devices were created when I added the mqtt server/user/pass in the aqualinkd, I didn’t have anything created in HA’s yaml.  All cleaned up and working now.  Appreciate the support!

Now it’s time to getting a HA view for a wall mounted kiosk to control the pool!