aqualinkd/AqualinkD #402: Upgrading to 2.6.3 causes iAqualink wifi connection to drop

brownjd, 2025-04-13:
Powercycling the RS8 panel will allow it to connect to wifi momentarily (the iAqualink IOS app shows green), but after aqualinkd establishes a connection, the wifi connection the panel drops (and iAqualink IOS app shows red and asks you to refresh). 

This behavior was not observed in 2.6.1/

sfeakes, 2025-04-13:
This sounds like you have `enable_iaqualink=YES` in your aqualinkd.conf.   Panel can only support one iaqualinkd device, so both Jandy's iAqualink wifi device and AqualinkD can't use that address.  Just set `enable_iaqualink=no` or comment the line out in aqualinkd.conf

brownjd, 2025-04-13:
You are correct. Setting device_id=0xFF for auto configure also seemed to enable_iaqualink=YES. I hadn't noticed that settings. Turning it off fixed the problem.

closed 2025-04-13 by brownjd

sfeakes, 2025-04-13:
Autoconfigure needs to be able to see all devices on the RS485 bus to configure correctly. Your iAqualink device may have been in sleep mode when AqualinkD ran through its autoconfigure.   Or It's a bug, I'll look into it.

brownjd, 2025-04-13:
I think it is because they are on the same breaker when I power cycled. They were both booting up at the same time. That may account for it.

sfeakes, 2025-04-13:
One thing I haven't fully documented yet, is if you do use 0xFF for auto configure, make sure you save the config AFTER it's finished self configuration. (seems like you have done that).  Otherwise it'll take a long time on next startup, and may overuse the `enable_iaqualink=NO` setting.