aqualinkd/AqualinkD #404: Docker - Seg fault launching /aqmanager.html

john-kapp, 2025-04-14:
First and foremost - thanks very much for a great project.  While I have a OneTouch and a PDA to control my pool, they are slow and clunky.  Your project has brought my pool control into the current decade.

I'm having one issue that I hope you may be able to shed some light on.  I've tried running the project in two different docker environments and while the main program works like a champ, whenever I try to switch into aqmanager by browsing to the http://aqualink.ip/aqmanager.html URL, the aqualinkd-docker process seg faults.  It appears that it is trying to write to journald which I suspect isn't available in Docker.  Are you aware of the issue and is there a workaround?  

`No such file or directory: Failed to open journal`
`/usr/local/bin/aqualinkd-docker: line 49:    21 Segmentation fault      (core dumped) /usr/local/bin/aqualinkd -d -c $AQUA_CONF`

Thanks very much.

sfeakes, 2025-04-14:
Are you using the supplied docker-compose.yaml ?
Have you looked at the docker section of the wiki?  
AqualinkD aqmanager depends on journal and the wiki shows you how to make that available to the container.

john-kapp, 2025-04-14:
Thanks for the tips.  I first tried running AqualinkD as a docker container in Unraid which doesn't appear to run journald.  I've just moved it over to running as a Docker addon under Home Assistant which does appear to run journald, but it imposes limitations on what directories can be mapped.  I'm likely going to abandon the idea of running this as a docker container and just create a small VM for it where I won't have to work around the platform's limitations.  Thanks again.

sfeakes, 2025-04-14:
Unraid is built on Slackware, so it does use journald / systemd. How you expose that to the container running on unraid I'm not sure.  I dislike HA's implementation of containers, it's horrible.  It's OK if you design specifically for deployment on HA, but deploying a generic container in HA is usually a disaster.  BUT, I know of a few people that deploy AqualinkD in a container on HA.
Personally I have a VM that manages all my "production" containers, using docker compose, since it's the most flexible and no UI's / anything else getting in the way.

BTW, it looks like Unraid has a docker compose plugin. That may be worth a shot, then use the docker-compose.yml supplied in this repo.  Also can't you use Unraid's Hostpath / Container Path to map the directory & socket needed for journald?

john-kapp, 2025-04-15:
While Unraid is based on Slackware, they add/remove packages as it serves their core purpose.  The system doesn't run journald out of the box.  There have been some Unraid community threads where people discuss adding it or one purpose or another, but most people conclude that it's typically better going a different route.  After all, Unraid is a NAS first and a Docker container host a distant second.

I did get the non-aquamanager pieces of AqualinkD working well under HA. But it would keep nagging me that I didn't having it working completely without aquamanager.

I now have a small Debian VM running Aqualink (on my Unraid system) and all components seem to be working well.  I think I'm going to declare victory with this.

Thanks again for a great application and your support.