aqualinkd/AqualinkD #408: LightMode 11 Default Brightness

brownjd, 2025-04-26:
Using 2.6.3 and LightMode 11, it is not clear what brightness it should default to when the button is clicked with a short press from the off setting. Currently, it does not turn on the lights, but simply reverts back to the off position. 

If the lights are on, you can successfully change the brightness using the aqualinkd web interface. However, if you use the web interface to turn them off (with a short click), you won't be able to turn them back on.

In fact, the only way to turn them on from the off setting, is by using the iAqualink interface. Of note, that interface does require you to set a brightness when you turn them on. 

I did try to turn them on using a long-press in the web interface (that would bring up the selection of duration and brightness), but that doesn't work either. 

Any thoughts?

sfeakes, 2025-04-26:
With a simple on/off click, AqualinkD will simply send the on/off to the panel. (ie no % brightness). The panel should turn on to the last % (ie 100% if there isn't one).
Obviously this isn't working in your case, let me see if I can reproduce the problem you are facing.

brownjd, 2025-04-26:
When you turn off the light in HomeKit, I think it sets the brightness to zero. That might be the problem.

Jason

sfeakes, 2025-05-10:
I made some changes in v2.6.4 that may help this.  Please try and let me know results.

brownjd, 2025-05-10:
Tested with 2.6.5. Clicking the light tile in the web view of auqalinkd, turns the tile on momentarily then back off once it polls and sees that the lights are not on. Bottom line, it does not turn on the lights.

Also, I can confirm that both in the IOS app and the web view of iAqualink (the zodiac site), if you just flip the switch for the light, it will prompt you for what brightness you want. You can't simply turn them on to the previous setting (it used to be this way, but not anymore). 

If the lights are on, clicking the tile will turn them off. Also, if the lights are on and you want to change the brightness, it works as expected. It is simply the off -> on behavior, regardless of whether it is a short or long press. Long press + selection of brightness does not work in aqualinkd interface.

sfeakes, 2025-05-10:
Can you let me know your panel rev, and post your aqualinkd.conf.   I can't seem to replicate this issue at all, providing that info may help.

sfeakes, 2025-05-10:
One other thing to try.  Make sure the light is off, then turn it on by running the following command on the Pi / machine running AqualinkD.  (Change `Aux_5` to the actual aux/button you have the light on)
```curl -X PUT -d value=40  "localhost/api/Aux_5/brightness/set"```

That will use a different command sequence, just wondering of that works.

brownjd, 2025-05-10:
> One other thing to try. Make sure the light is off, then turn it on by running the following command on the Pi / machine running AqualinkD. (Change `Aux_5` to the actual aux/button you have the [...]

pi@aqualink:~ $ curl -v -X PUT -d value=40 "localhost/api/Aux_2/brightness/set"
*   Trying ::1:80...
* connect to ::1 port 80 failed: Connection refused
*   Trying 127.0.0.1:80...
* Connected to localhost (127.0.0.1) port 80 (#0)
> PUT /api/Aux_2/brightness/set HTTP/1.1
> Host: localhost
> User-Agent: curl/7.74.0
> Accept: */*
> Content-Length: 8
> Content-Type: application/x-www-form-urlencoded
> 
* upload completely sent off: 8 out of 8 bytes
* Mark bundle as not supporting multiuse
< HTTP/1.1 400 Bad Request
< Server: Mongoose/6.18
< Content-Type: text/plain
< Content-Length: 15
< 
* Connection #0 to host localhost left intact
Unknown command

brownjd, 2025-05-10:
> Can you let me know your panel rev, and post your aqualinkd.conf. I can't seem to replicate this issue at all, providing that info may help.

Aqualinkd 2.6.5
B0029226 REV T.2
RS-8 Only

----
socket_port=80
serial_port=/dev/ttyUSB0

log_level=NOTICE

web_directory=/var/www/aqualinkd/

panel_type=RS-8 Only (Pool or Spa)

device_id=0x08
rssa_device_id=0x48
extended_device_id=0x30
enable_iaqualink=NO

mqtt_address=localhost:1883
#mqtt_user=
#mqtt_passwd=
mqtt_aq_topic=aqualinkd
mqtt_ha_discover_topic=homeassistant
mqtt_ha_discover_use_mac=YES
mqtt_timed_update=YES
mqtt_convert_temp_to_c=YES
#mqtt_dz_sub_topic=
#mqtt_dz_pub_topic=

#dzidx_air_temp=
#dzidx_pool_water_temp=
#dzidx_spa_water_temp=
#dzidx_SWG_percent=
#dzidx_SWG_PPM=
#dzidx_SWG_Status=
dz_convert_temp_to_c=YES

light_programming_mode=0.001000
light_programming_initial_on=0
light_programming_initial_off=0

read_RS485_swg=NO
read_RS485_ePump=YES
read_RS485_vsfPump=NO
read_RS485_JXi=NO
read_RS485_LX=NO
read_RS485_Chem=NO
read_RS485_iAqualink=NO
read_RS485_HeatPump=NO

force_swg=NO
force_ps_setpoints=NO
force_frzprotect_setpoints=YES
force_chem_feeder=NO
force_chiller=NO

enable_scheduler=NO
event_check_use_scheduler_times=NO
event_poweron_check_pump=NO
event_freezeprotectoff_check_pump=NO
event_boostoff_check_pump=NO
event_check_pumpon_hour=0
event_check_pumpoff_hour=0
#event_booston_check_device=

ftdi_low_latency=YES

rs485_frame_delay=0

sync_panel_time=YES

display_warnings_in_web=YES

override_freeze_protect=NO

report_zero_spa_temp=NO
report_zero_pool_temp=NO

pda_sleep_mode=YES

button_01_label=Filter Pump
button_01_pumpIndex=1
button_01_pumpID=0x78
button_01_pumpType=JANDY ePUMP

button_02_label=Cleaner

button_03_label=Pool Lights
button_03_lightMode=11

button_04_label=Sheer Descent

button_05_label=NONE

button_06_label=NONE

button_07_label=NONE

button_08_label=NONE

button_09_label=Pool Heater

button_10_label=NONE

button_11_label=NONE

sfeakes, 2025-05-10:
Are you running in a docker?  
AqualinkD will not refuse a command on HTTP, so either you are not on the local machine when you ran that command, or you are not using the default port 80.

> pi@aqualink:~ $ curl -v -X PUT -d value=40 "localhost/api/Aux_2/brightness/set"
>
> * Trying ::1:80...
> * connect to ::1 port 80 failed: Connection refused
> * Trying 127.0.0.1:80...
> * Connected to [...]

You can ignore the reply. AqualinkD should have tried to execute the command.  If you go to aqmanager and turn on `NetServices` under `DebugMasks` you should see AqualinkD received the command and tried to execute it.
So I'm assuming it didn't turn the light on.

sfeakes, 2025-05-10:
Thanks for posting the config.  Let me see if I can replicate the issue on a T.2 panel.

brownjd, 2025-05-10:
switched to using the hostname just to demonstrate. See the second command below. 

curl -v -X PUT -d value=40 "aqualink.local/api/Aux_2/brightness/set"
*   Trying 192.168.86.232:80...
* Connected to aqualink.local (192.168.86.232) port 80 (#0)
> PUT /api/Aux_2/brightness/set HTTP/1.1
> Host: aqualink.local
> User-Agent: curl/7.74.0
> Accept: */*
> Content-Length: 8
> Content-Type: application/x-www-form-urlencoded
> 
* upload completely sent off: 8 out of 8 bytes
* Mark bundle as not supporting multiuse
< HTTP/1.1 400 Bad Request
< Server: Mongoose/6.18
< Content-Type: text/plain
< Content-Length: 15
< 
* Connection #0 to host aqualink.local left intact
Unknown command

curl aqualink.local | head -n 10
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0<!DOCTYPE html>
<html lang='en'>

<head>
  <meta http-equiv='Content-Type' content='text/html; charset=windows-1252'>
  <title>AqualinkD</title>
  <meta name='viewport' content='width=device-width'>
  <meta name='mobile-web-app-capable' content='yes'>
  <meta name='apple-mobile-web-app-capable' content='yes'>
  <meta name='apple-mobile-web-app-status-bar-style' content='black'>
 28  116k   28 33933    0     0   424k      0 --:--:-- --:--:-- --:--:--  430k

brownjd, 2025-05-10:
> Are you running in a docker? AqualinkD will not refuse a command on HTTP, so either you are not on the local machine when you ran that command, or you are not using the default port 80.

No. I ran the command from the host. I reran using the hostname instead of localhost. See the other comment.

> You can ignore the reply. AqualinkD should have tried to execute the command. If you go to aqmanager and turn on `NetServices` under `DebugMasks` you should see AqualinkD received the command and tried to execute it. So I'm assuming it didn't turn the light on.

It did not. But I think the command is running.

Debug: NetService:Started AqualinkD Manager
Debug: NetService:API: URI Request 'Aux_2/brightness/set': value 40.00
Info:  NetService:Message body:'value=40'
Info:  NetService:URI request: : '/api/Aux_2/brightness/set '

sfeakes, 2025-05-10:
Thanks for confirming.   This may take longer to debug, I was hoping the command sequence AqualinkD would have run using the curl command would have worked.

brownjd, 2025-06-11:
I've been noticing that the iAqualink setting that allows you to set whether you are using a dimmer seems to randomly revert to non-dimmable. 

It never happened before attempting to use the new Lightmode 11. I mention it because any time I want to test the dimmer (which sometimes works if the light is already on), I need to periodically reset this panel setting.

[image]

sfeakes, 2025-06-16:
In HomeKit click the icon and not the button, that will turn the dimmer on / off without having to use the % slider.

Either way, with the latest software, setting the dimmer to 0 is the same as clicking the icon / turning it off directly as long as you move the % quickly enough.

I'm still looking into why this is happening for you, but still can't replicate it.

brownjd, 2025-06-17:
I use Home.app scenes to set the brightness at different times.

Sunset -> 100%
10 pm -> 33%
12 pm -> Off

The current behavior is that the lights some on at Sunset at 33%.

Now, if I manually set the brightness or turn them off by setting the brightness to 0, all bets are off. At sunset, they’ll probably stay off.

I’ve noticed that the iAqualink web interface behavior is such that when you turn it on and it prompts you for the brightness, it turns them on to 100% before you make your selection (then it sets them to what you selected).

Perhaps there is missing step here.

Jason

sfeakes, 2025-06-17:
> browned
> I've been playing with it some more and I've noticed that if you turn off the lights using IOS Home, you **have** to slide down the brightness to 0 (there's no ability to click off).

> sfeakes
> In HomeKit click the icon and not the button, that will turn the dimmer on / off without having to use the % slider.

Did you try this ?  In the previous reply you were saying that sliding the slider to 0 to turn off caused the problem, and that's the only way you can turn them off.    You DON'T have to do that in the Home.app, you can turn them off without touching the % slider.

BTW, AqualinkD sets the brightness to 100 to turn them on, then sets to the request %.  It just does it a LOT faster than iAqualink.

brownjd, 2025-06-17:
So you said HomeKit, but I think you meant Homebridge. I tested it repeatedly through its web UI and while it seemed to work, after several cycles it stops cycling the lights and incorrectly reports that they are on when they aren’t.

In the IOS Home.app (HomeKit) there is no way to turn on a dimmable switch without first setting the brightness.

This is also the behavior of the web and IOS versions of iAquaLink.

The behavior is fairly inconsistent. I tested several times and I can also get it into a state where Homebridge reports the lights are off when they are on.

I then decided to eliminate the middleman and went straight to the Aqualinkd web UI. Same behavior. Once it gets into a bad state, the lights stop responding. It doesn’t seem to suffer from misreporting the state, but it will eventually lose the ability to cycle the lights.

The only way to clear the condition is to cycle the lights with the iAquaLink interface.

Jason

brownjd, 2025-06-18:
Same issue. After repeated cycling and playing with brightness you lose the ability to cycle. Clicking the bulb icon sets it to the previous brightness but then reverts to off after piolling. If you try this several times, it will eventually just stay lit even through the lights are off.

Jason

brownjd, 2025-09-01:
> @brownjd I think I've finally found the issue. Looks like Jandy Panel Rev T.2 (maybe others) has a bug for the light dimmers when using the RS serial protocol. AqualinkD can now detect a "stuck" light, reset it and continue as normal. It will be slightly slower turning ON a light than previously, and since it now uses AllButton protocol to turn on the light, it will only turn on to 25/50/75/100, then change to selected value if not one of those 25% increments.

Thanks!

I did some limited testing with HomeKit scenes that automate turning them on and off with various brightness settings.

Short version - way more reliable. I’ll keep testing, but so far so good. I am able to set the brightness outside of the 25% increments using HomeKit, 77% for example.

The turn on switch vs specifying the brightness behavior does not always seem to pick up the last setting, but I’ll keep testing to make sure.

How did you know about the hardware bug without being able to test on any hardware?

Jason

sfeakes, 2025-09-01:
Thanks, the code needs a LOT of cleanup, but wanted to see if this fixed it for you first.
When it doesn't pick the last brightness, look in the log for the following line.
`AllButton: Light Programming detected Jandy panel light% bug, re-setting to 100%`
It may have had to re-set the light, if which case it will loose the last setting and defaults to 100%
If that's the case, I could make it default to the closest 25% increment to the last setting if that's in AqualinkD's history (ie AqualinkD hasn't been restarted).

sfeakes, 2025-09-04:
From my diagnosis The bug seems to be with the Jandy panel firmware rev T (maybe some others). So not surprising iAqualink has a similar issue.

When this happens from iAqualink, if you turn the light on with AqualinkD see if it detects the light is stuck and fixes it for you.  I believe it will.

- Yes AqualinkD simulates wired controllers, look at the simulator page.

- BTW I don’t think you don’t need to go through the steps you mentioned to reset the light, the Allbutton protocol can do that quicker and easier. (Press the light button-> should display 0 or negative number if stuck-> press left or right to get the % you want -> hit enter.   That should reset far quicker than the method you mentioned.

brownjd, 2025-09-04:
If the firmware loses the setting indicating there is a dimmer, then the lights won’t respond from any interface, including iAqualink. In fact, iAqualink will incorrectly report that the light is on (with no brightness number).

Below you mention resetting the light. Do you mean that by just setting the brightness through Aqualinkd or HomeKit, it will clear the flag and set it to having dimmer again?

brownjd, 2025-09-04:
> Yes. If the light is stuck through iAqualink, then try Turing it on through AqualinkD. AqualinkD will detect the light is stuck and re-set it. It should then work through iAqualink again.
> BTW, From my testing the Jandy Control Panel firmware hasn't lost the light or setting, it has the light stuck in an invalid mode. To get over that, you can delete the light and re-add it (as you described in this post) , or I found a quicker way using the AllButton protocol that will do a lower level reset of the light that's quicker. However the light is reset, it will loose the last setting, and come back on at 100%, BUT at least it will work.
> Since you have AqualinkD and HomeKit, I'm not sure why you are even using iAqualink, what is that providing that AqualinkD with HomeKit can't?

So I power cycled the board and the setting did revert back to no dimmer. I know you said it is in an invalid state, but see the attached screenshot. At least in the iAqualink setup menu, it no longer has an X by Pool Light.

I tried turning on the light from aqualinkd, and it does not turn on the lights. Both iAqualink and aqualinkd think the lights are on, but they aren’t. Aqualinkd thinks the brightness is 0.

Here is the log output.

Notice:Panl&Prog: Starting programming thread 'Set light dimmer'
Notice:Panl&Prog: Starting programming thread 'Set light dimmer'
Error: AllButton: Light Programming didn't receive light mode message
Error: AllButton: Light Programming didn't receive light mode message

Now, you asked why I use iAqualink - because not everything has been working through aqualinkd. It is my backup. I also still use it for scheduling the filter, pool cleaner, etc.

sfeakes, 2025-09-04:
Power cycling the panel when the light is stuck, is something I have not tested, so I have no idea on the outcome. If the light does get deleted from the config when you do that, then AqualinkD will not be able to turn in on, unless you re-configure AqualinkD and delete the light. ie match AqualinkD's config to the panel's config.
The output you posted from AqualinkD is what I would expect when the configurations do not match.

The next time you notice iAqualink not being able to turn on the light, see if AqualnkD can turn it on.  If it does, then my bet is iAqualink will also be able to control it again.

You can also manually re-set the light using AqualinkD's OneTouch simulator and follow the steps you mentioned.  ie go into settings, delete the light and re-create the light.  Or you can use AqualinkD's allbutton simulator and follow the steps I previously mentioned.  Both ways will re-set the light back to working in the tests that I have done on the only panel rev I found with the issue (Rev T).    BTW, what panel rev# do you have?

brownjd, 2025-09-06:
Just for giggles, I flipped the breaker to reboot both the board and aqualinkd. I wanted to be sure the dimmer flag stayed set. 

Sadly, it did not. I thought we had it.

I tried going through one touch simulator to set the dimmer again. Here is the output.

Notice the message at the bottom in bold. Is that from setting of the flag or polling of the light state?

Notice:AqualinkD: Starting Aqualink Daemon v2.6.10 !
Notice:AqualinkD: socket port                         = 80
Notice:AqualinkD: serial port                         = /dev/ttyUSB0
Notice:AqualinkD: log level                           = NOTICE
Notice:AqualinkD: web directory                       = /var/www/aqualinkd/
Notice:AqualinkD: panel type                          = RS-8 Only (Pool or Spa)
Notice:AqualinkD: device id                           = 0x08
Notice:AqualinkD: rssa device id                      = 0x48
Notice:AqualinkD: extended device id programming      = YES
Notice:AqualinkD: extended device id                  = 0x30
Notice:AqualinkD: enable iaqualink                    = NO
Notice:AqualinkD: mqtt address                        = localhost:1883
Notice:AqualinkD: mqtt user                           =
Notice:AqualinkD: mqtt passwd                         =
Notice:AqualinkD: mqtt aq topic                       = aqualinkd
Notice:AqualinkD: mqtt ha discover topic              = homeassistant
Notice:AqualinkD: mqtt ha discover use mac            = YES
Notice:AqualinkD: mqtt timed update                   = YES
Notice:AqualinkD: mqtt convert temp to c              = YES
Notice:AqualinkD: light programming mode              = 0.001000
Notice:AqualinkD: light programming initial on        = 0
Notice:AqualinkD: light programming initial off       = 0
Notice:AqualinkD: read RS485 swg                      = NO
Notice:AqualinkD: read RS485 ePump                    = YES
Notice:AqualinkD: read RS485 vsfPump                  = NO
Notice:AqualinkD: read RS485 JXi                      = NO
Notice:AqualinkD: read RS485 LX                       = NO
Notice:AqualinkD: read RS485 Chem                     = NO
Notice:AqualinkD: read RS485 iAqualink                = NO
Notice:AqualinkD: read RS485 HeatPump                 = NO
Notice:AqualinkD: force swg                           = NO
Notice:AqualinkD: force ps setpoints                  = NO
Notice:AqualinkD: force frzprotect setpoints          = YES
Notice:AqualinkD: force chem feeder                   = NO
Notice:AqualinkD: force chiller                       = NO
Notice:AqualinkD: enable scheduler                    = NO
Notice:AqualinkD: event check use scheduler times     = NO
Notice:AqualinkD: event poweron check pump            = NO
Notice:AqualinkD: event freezeprotectoff check pump   = NO
Notice:AqualinkD: event boostoff check pump           = NO
Notice:AqualinkD: event check pumpon hour             = 0
Notice:AqualinkD: event check pumpoff hour            = 0
Notice:AqualinkD: event booston check device          =
Notice:AqualinkD: ftdi low latency                    = YES
Notice:AqualinkD: rs485 frame delay                   = 0
Notice:AqualinkD: sync panel time                     = YES
Notice:AqualinkD: display warnings in web             = YES
Notice:AqualinkD: override freeze protect             = NO
Notice:AqualinkD: report zero spa temp                = NO
Notice:AqualinkD: report zero pool temp               = NO
Notice:AqualinkD: pda sleep mode                      = YES
Notice:AqualinkD: sensor poll time                    = 300
Notice:AqualinkD: save debug log masks                = NO
Notice:AqualinkD: save light programming value        = NO
Notice:AqualinkD: Button Filter_Pump   = label Filter Pump     | VSP ID 0x78 | PumpID 1 |
Notice:AqualinkD: Button Aux_1         = label Cleaner         |
Notice:AqualinkD: Button Aux_2         = label Lights          | Light Progm | CTYPE 11  |
Notice:AqualinkD: Button Aux_3         = label Sheer Descent   |
Notice:AqualinkD: Button Aux_4         = label NONE            |
Notice:AqualinkD: Button Aux_5         = label NONE            |
Notice:AqualinkD: Button Aux_6         = label NONE            |
Notice:AqualinkD: Button Aux_7         = label NONE            |
Notice:AqualinkD: Button Pool_Heater   = label Heater          |
Notice:AqualinkD: Button Spa_Heater    = label NONE            |
Notice:AqualinkD: Button Extra_Aux     = label NONE            |
Notice:NetService:Starting network services thread
Notice:RS Serial: Port /dev/ttyUSB0 low latency mode is NOT set, resetting to low latency!
Notice:AqualinkD: Listening to Aqualink RS-8 Only (Pool or Spa) on serial port: /dev/ttyUSB0
Notice:AqualinkD: Waiting for Control Panel probe
Notice:NetService:Starting web server on port 80
Notice:NetService:Starting MQTT client to localhost:1883
Notice:AqualinkD: Got probe on '0x48' RS SerialAdapter Protocol
Notice:AqualinkD: Got probe on '0x08' Standard Protocol
Notice:AqualinkD: Got probe on '0x30' Extended Protocol
Notice:Panl&Prog: AqualinkD is using use IAQ TOUCH mode for programming (where supported)
Notice:AqualinkD: Starting communication with Control Panel
Notice:Panl&Prog: Starting programming thread 'Get SerialAdapter setpoints'
Notice:Panl&Prog: Panel REV T.2 from AllButton
Notice:Panl&Prog: Panel CPU B0029226 from AllButton
Notice:Panl&Prog: Starting programming thread 'Get Freeze protect'
Notice:Simulator: Starting AllButton Simulator!
Notice:Simulator: Got probe on '0x09', using for simulator ID
Notice:Simulator: Starting OneTouch Simulator!
Notice:Panl&Prog: Starting programming thread 'Set light dimmer'
Notice:Simulator: Starting OneTouch Simulator!
Notice:Simulator: Starting AllButton Simulator!
Notice:Simulator: Got probe on '0x09', using for simulator ID
**Error: AllButton: Light Programming didn't receive light mode message**
Notice:Simulator: Starting OneTouch Simulator!
Notice:Simulator: Got probe on '0x43', using for simulator ID

sfeakes, 2025-09-15:
I have not messed around with HomeKit scenes / automation, after the light has turned off (from the 50%) setting, 
Can you try turning it on with 100% from the web UI?  
I’m wondering if the scene (or HomeKit in general) is simply sending the on command, or on to 100% command, (the two are different).   I’ll play with it when I get some time.

brownjd, 2025-09-17:
> If you enable "Panel&Prog" and/or "NetServices" under debug masks, that will show you exactly what's being sent. My guess is it's sending the "on" command then hopefully "100%" command after.
> AqualinkD will act immediately on the "on" command, but with "%" it will weight a second before acting on the command since HomeKit will throw lots of % messages as you increase and decrease the slider, and obviously we only want the last command.
> HomeKit is designed around LED lights that can act immediately, not something like the Jandy panel that's very slow. HomeAssistant (and some other automation hubs) on the other hand will only throw the % message once the user has finished with the slider.

I shut off the lights (last brightness 50%) using the go to bed scene. Lights turned off.
Then I turned on the debug masks.
The output below was generated from activating the sunset scene, which should turn the lights to 100. Instead, they end up at 50%.

ebug: NetService:MQTT: Ignore aqualinkd/Filter_Pump/Watts 1000
Debug: NetService:MQTT: Message publishing acknowledged (msg_id: 41782)
Debug: NetService:WS: Message - Key 'uri' Value 'removelogmask'
Debug: NetService:WS: Message - Key 'value' Value '1024'
Debug: NetService:WS: URI Request 'removelogmask': value 1024.00
Debug: NetService:Started AqualinkD Manager
Debug: NetService:WS: Message - Key 'uri' Value 'removelogmask'
Debug: NetService:WS: Message - Key 'value' Value '2'
Debug: NetService:WS: URI Request 'removelogmask': value 2.00
Debug: NetService:Started AqualinkD Manager
Debug: NetService:WS: Message - Key 'uri' Value 'addlogmask'
Debug: NetService:WS: Message - Key 'value' Value '256'
Debug: NetService:WS: URI Request 'addlogmask': value 256.00
Debug: NetService:Started AqualinkD Manager
Debug: NetService:MQTT: topic aqualinkd/Aux_2/set 1
Debug: NetService:MQTT: URI Request 'Aux_2/set': value 1.00
Info:  NetService:0: MATCH Aux_2 to topic Aux_2/set
Info:  NetService:ri1=Aux_2/set1tatu! , length=6 char at len=s
Notice:Panl&Prog: Starting programming thread 'Set light dimmer'
Info:  NetService:MQTT: Published id=41786: aqualinkd/Display_Message Programming: setting light mode
Debug: NetService:MQTT: Ignore aqualinkd/Display_Message Programming: setting light mode
Debug: NetService:MQTT: Message publishing acknowledged (msg_id: 41786)
Info:  NetService:MQTT: Published id=41787: aqualinkd/Aux_2/delay 0
Info:  NetService:MQTT: Published id=41788: aqualinkd/Aux_2 1
Info:  NetService:MQTT: Published id=41789: aqualinkd/Aux_2/timer 0
Info:  NetService:MQTT: Published id=41790: aqualinkd/Aux_2/timer/duration 0
Info:  NetService:MQTT: Published id=41791: aqualinkd/Aux_2/program 50
Info:  NetService:MQTT: Published id=41792: aqualinkd/Aux_2/program/name 50%
Info:  NetService:MQTT: Published id=41793: aqualinkd/Aux_2/brightness 50
Debug: NetService:MQTT: Ignore aqualinkd/Aux_2/delay 0
Debug: NetService:MQTT: Message publishing acknowledged (msg_id: 41787)
Debug: NetService:MQTT: received (msg_id: 41787), looks like my own message, ignoring
Debug: NetService:MQTT: Ignore aqualinkd/Aux_2 1
Debug: NetService:MQTT: Message publishing acknowledged (msg_id: 41788)
Debug: NetService:MQTT: received (msg_id: 41788), looks like my own message, ignoring
Debug: NetService:MQTT: Ignore aqualinkd/Aux_2/timer 0
Debug: NetService:MQTT: Message publishing acknowledged (msg_id: 41789)
Debug: NetService:MQTT: received (msg_id: 41789), looks like my own message, ignoring
Debug: NetService:MQTT: Ignore aqualinkd/Aux_2/timer/duration 0
Debug: NetService:MQTT: Message publishing acknowledged (msg_id: 41790)
Debug: NetService:MQTT: received (msg_id: 41790), looks like my own message, ignoring
Debug: NetService:MQTT: Ignore aqualinkd/Aux_2/program 50
Debug: NetService:MQTT: Message publishing acknowledged (msg_id: 41791)
Debug: NetService:MQTT: received (msg_id: 41791), looks like my own message, ignoring
Debug: NetService:MQTT: Ignore aqualinkd/Aux_2/program/name 50%
Debug: NetService:MQTT: Message publishing acknowledged (msg_id: 41792)
Debug: NetService:MQTT: received (msg_id: 41792), looks like my own message, ignoring
Debug: NetService:MQTT: Ignore aqualinkd/Aux_2/brightness 50
Debug: NetService:MQTT: Message publishing acknowledged (msg_id: 41793)
Info:  NetService:MQTT: Published id=41794: aqualinkd/Filter_Pump/Watts 1000
Info:  NetService:MQTT: Published id=41795: aqualinkd/Aux_2/program 100
Info:  NetService:MQTT: Published id=41796: aqualinkd/Aux_2/program/name 100%
Info:  NetService:MQTT: Published id=41797: aqualinkd/Aux_2/brightness 100
Debug: NetService:MQTT: Ignore aqualinkd/Filter_Pump/Watts 1000
Debug: NetService:MQTT: Message publishing acknowledged (msg_id: 41794)
Debug: NetService:MQTT: received (msg_id: 41794), looks like my own message, ignoring
Debug: NetService:MQTT: Ignore aqualinkd/Aux_2/program 100
Debug: NetService:MQTT: Message publishing acknowledged (msg_id: 41795)
Debug: NetService:MQTT: received (msg_id: 41795), looks like my own message, ignoring
Debug: NetService:MQTT: Ignore aqualinkd/Aux_2/program/name 100%
Debug: NetService:MQTT: Message publishing acknowledged (msg_id: 41796)
Debug: NetService:MQTT: received (msg_id: 41796), looks like my own message, ignoring
Debug: NetService:MQTT: Ignore aqualinkd/Aux_2/brightness 100
Debug: NetService:MQTT: Message publishing acknowledged (msg_id: 41797)
Info:  NetService:MQTT: Published id=41798: aqualinkd/Display_Message
Debug: NetService:MQTT: Ignore aqualinkd/Display_Message
Debug: NetService:MQTT: Message publishing acknowledged (msg_id: 41798)
Info:  NetService:MQTT: Published id=41799: aqualinkd/Aux_2/program 50
Info:  NetService:MQTT: Published id=41800: aqualinkd/Aux_2/program/name 50%
Info:  NetService:MQTT: Published id=41801: aqualinkd/Aux_2/brightness 50
Debug: NetService:MQTT: Ignore aqualinkd/Aux_2/program 50
Debug: NetService:MQTT: Message publishing acknowledged (msg_id: 41799)
Debug: NetService:MQTT: received (msg_id: 41799), looks like my own message, ignoring
Debug: NetService:MQTT: Ignore aqualinkd/Aux_2/program/name 50%
Debug: NetService:MQTT: Message publishing acknowledged (msg_id: 41800)
Debug: NetService:MQTT: received (msg_id: 41800), looks like my own message, ignoring
Debug: NetService:MQTT: Ignore aqualinkd/Aux_2/brightness 50
Debug: NetService:MQTT: Message publishing acknowledged (msg_id: 41801)
Debug: NetService:WS: Message - Key 'uri' Value 'removelogmask'
Debug: NetService:WS: Message - Key 'value' Value '2'
Debug: NetService:WS: URI Request 'removelogmask': value 2.00

Jason

sfeakes, 2025-09-17:
In that log, there was only one request made to AqualinkD, and that was to turn on the light at the last setting.
Debug: NetService:MQTT: URI Request 'Aux_2/set': value 1.00

To set the %, you would see a URI request like Aux_2/brightness/set: value 100.  

I’ll have to play with HomeKit scenes when I get chance.

brownjd, 2025-09-19:
> I will try turning on logging in HomeBridge to see what it gets from the AppleTV when a scene is activated.
>
> Jason

Is there a debug logging option for the aqualinkd plugin? Turning it on for homebridge did not enable anything helpful in the plugin.

brownjd, 2026-03-12:
This works if you set HomeKit to turn on the lights to 99% brightness. If you select 100%, it doesn’t always work.