aqualinkd/AqualinkD #412: Installed AquaLinkD 2.6.4 and Virtual Buttons Are Not All Showing

jkosharek, 2025-05-06:
Installed AquaLinkD 2.6.4 and Virtual Buttons Are Not All Showing on the Web UI or in Home Assistant. 

OneTouches on the iAqualink side:
1: All Off
2: Spa On
3: Clean On
4: SP1
5: SP2
6: SP3

After installing AquaLinkD 2.6.4, 1-4 showed up on the AquaLinkD Web UI and in Home Assistant but 5 and 6 did not. Labels match the OneTouches in iAqualink (if that matters). 
```
virtual_button_01_label=All Off
virtual_button_01_onetouchID=1
virtual_button_02_label=Spa On
virtual_button_02_onetouchID=2
virtual_button_03_label=Clean On
virtual_button_03_onetouchID=3
virtual_button_04_label=SP1
virtual_button_04_onetouchID=4
virtual_button_05_label=SP2
virtual_button_05_onetouchID=5
virtual_button_06_label=SP3
virtual_button_06_onetouchID=6
```

jkosharek, 2025-05-06:
It is the RS16

sfeakes, 2025-05-10:
I just posted v2.6.5 that should fix this.

jkosharek, 2025-05-10:
Yes it did nice work! With the RS-16, what would be the max number of virtual buttons?

closed 2025-05-10 by jkosharek

sfeakes, 2025-05-10:
It's currently 8 for an RS16.   I can increase if needed.

28 is the total number of buttons in AqualinkD. 
RS16 panel needs 20 buttons for its inbuilt circuits (and heaters / jva etc), leaving 8 for virtual.
So RS8 would have 16 virtual buttons available.