aqualinkd/AqualinkD #413: PDA Device programmer couldn't find menu item on any page 'Pool Heater   '

microbmen, 2025-05-14:
Hello, 

I have a PDA-PS8 Combo panel,  I can control my heater via the PDA panel, and the AqualinkD Panel simulator.   However when I use the AqualinkD Default Web Interface I cannot turn the heater on or off.  I can however change the temperature set point.

I get the follow error:

`
May 13 22:47:09 Info:   NetService:2: MATCH Pool_Heater to topic Pool_Heater/set
May 13 22:47:09 Info:   NetService:ri1=Pool_Heater/set, length=11 char at len=/
May 13 22:47:09 Info:   Panl&Prog: Device request type 'OnOff' for deviceindex 9 'Pool Heater' of value 0 from 'WebSocket'
May 13 22:47:09 Info:   Panl&Prog: received 'OFF' for 'Pool_Heater', turning 'OFF'
May 13 22:47:09 Notice: Panl&Prog: Starting programming thread 'Switch PDA device on/off'
May 13 22:47:09 Info:   Panl&Prog: Programming: Switch PDA device on/off, 19
May 13 22:47:09 Info:   PDA:       PDA Device On/Off, device 'Pool Heater', state 1
May 13 22:47:09 Info:   AllButton: Queue send '0x02' to controller (programming)
May 13 22:47:09 Info:   NetService:MQTT: Published id=309: aqualinkd/Display_Message Programming: setting device on/off
May 13 22:47:09 Info:   NetService:MQTT: Published id=310: aqualinkd/Filter_Pump/delay 1
May 13 22:47:09 Info:   NetService:MQTT: Published id=311: aqualinkd/Filter_Pump 1
May 13 22:47:09 Info:   NetService:MQTT: Published id=312: aqualinkd/Filter_Pump/timer 0
May 13 22:47:09 Info:   NetService:MQTT: Published id=313: aqualinkd/Filter_Pump/timer/duration 0
May 13 22:47:09 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:11 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:11 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:11 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:11 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:11 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:11 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:11 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:12 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:12 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:13 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:13 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:14 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:14 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:14 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:15 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:15 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:16 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:16 Info:   AllButton: Queue send '0x05' to controller (programming)
May 13 22:47:16 Info:   AllButton: Queue send '0x05' to controller (programming)
**May 13 22:47:16 Error:  PDA:       PDA Device programmer couldn't find menu item on any page 'Pool Heater   '
May 13 22:47:16 Error:  PDA:       PDA Device On/Off, device 'Pool Heater' not found**
May 13 22:47:18 Info:   NetService:MQTT: Published id=314: aqualinkd/Display_Message

`

I have a JXi heater plugged in via RS485.  

It seems to me like a parser issue and not finding the heater menu item for this model panel...  Any help is appriciated.

microbmen, 2025-05-14:
System Information
AqualinkD version:	2.6.5
Panel version:	PDA-PS8 Combo PDA: 7.2
Panel type:  PDA-8 Combo

microbmen, 2025-05-14:
Here's a bit more logs from debug,  its seems to me like I found a bug.  It is cycling through menus looking for 'Pool Heater', but in this version, it must be 'Pool Heat'.   

Here are additional logs... .

May 14 06:50:54 Info:   AllButton: Queue send '0x05' to controller (programming)
May 14 06:50:54 Debug:  PDA:       waitForPDAMessageTypes  0x08 or 0x04
May 14 06:50:54 Debug:  PDA:       waitForPDAMessageTypes 0x08 | 0x04, last message type was 0x02 (1 of 8)
May 14 06:50:54 Debug:  RS Timings:Time between packets (0.028 sec)
May 14 06:50:54 Debug:  PDA:
May 14 06:50:54 Info:   PDA:       PDA Menu Line 0 =    EQUIPMENT
May 14 06:50:54 Info:   PDA:       PDA Menu Line 1 = AUX1         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 2 = AUX2         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 3 = AUX3         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 4 = AUX4         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 5 = AUX5         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 6 = AUX6         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 7 = AUX7         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 8 =
May 14 06:50:54 Info:   PDA:       PDA Menu Line 9 =    ^^ MORE __
May 14 06:50:54 Debug:  PDA:       PDA Menu highlighted line#=8 line=''
May 14 06:50:54 Debug:  PDA:       Kicking PDA thread 19,0xb6113c00
May 14 06:50:54 Debug:  PDA:       waitForPDAMessageTypes 0x08 | 0x04, last message type was 0x0f (2 of 8)
May 14 06:50:54 Debug:  PDA:       PDA can't sleep as thread 19,0xb6113c00 is active
May 14 06:50:54 Debug:  RS Timings:Time between packets (0.042 sec)
May 14 06:50:54 Info:   PDA:       PDA Menu Line 0 =    EQUIPMENT
May 14 06:50:54 Info:   PDA:       PDA Menu Line 1 = AUX1         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 2 = AUX2         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 3 = AUX3         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 4 = AUX4         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 5 = AUX5         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 6 = AUX6         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 7 = AUX7         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 8 = EXTRA AUX    OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 9 =    ^^ MORE __
May 14 06:50:54 Debug:  PDA:       PDA Menu highlighted line#=8 line='EXTRA AUX    OFF'
May 14 06:50:54 Debug:  PDA:       *** Checking Equiptment 'EXTRA AUX   '
May 14 06:50:54 Debug:  PDA:       *** Found EQ CTL Status for Extra Aux = 'EXTRA AUX    OFF'
May 14 06:50:54 Debug:  PDA:       Kicking PDA thread 19,0xb6113c00
May 14 06:50:54 Debug:  PDA:       waitForPDAMessageTypes 0x08 | 0x04, last message type was 0x04 (3 of 8)
May 14 06:50:54 Debug:  PDA:       waitForPDAMessageTypes: received 0x04
May 14 06:50:54 Debug:  AllButton: Waiting for queue to empty
May 14 06:50:54 Debug:  PDA:       PDA can't sleep as thread 19,0xb6113c00 is active
May 14 06:50:54 Debug:  RS Timings:Time between packets (0.028 sec)
May 14 06:50:54 Debug:  RS Timings:Time between packets (0.055 sec)
May 14 06:50:54 Debug:  RS Timings:Time between packets (0.056 sec)
May 14 06:50:54 Debug:  RS Timings:Time between packets (0.059 sec)
May 14 06:50:54 Debug:  RS Timings:Time between packets (0.055 sec)
May 14 06:50:54 Debug:  RS Timings:Time between packets (0.061 sec)
May 14 06:50:54 Debug:  PDA:       Kicking PDA thread 19,0xb6113c00
May 14 06:50:54 Debug:  PDA:       PDA can't sleep as thread 19,0xb6113c00 is active
May 14 06:50:54 Debug:  AllButton: Queue now empty!
May 14 06:50:54 Info:   AllButton: Queue send '0x05' to controller (programming)
May 14 06:50:54 Debug:  PDA:       waitForPDAMessageTypes  0x08 or 0x04
May 14 06:50:54 Debug:  PDA:       waitForPDAMessageTypes 0x08 | 0x04, last message type was 0x02 (1 of 8)
May 14 06:50:54 Debug:  RS Timings:Time between packets (0.023 sec)
May 14 06:50:54 Debug:  PDA:
May 14 06:50:54 Info:   PDA:       PDA Menu Line 0 =    EQUIPMENT
May 14 06:50:54 Info:   PDA:       PDA Menu Line 1 = AUX2         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 2 = AUX3         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 3 = AUX4         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 4 = AUX5         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 5 = AUX6         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 6 = AUX7         OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 7 = EXTRA AUX    OFF
May 14 06:50:54 Info:   PDA:       PDA Menu Line 8 =
May 14 06:50:54 Info:   PDA:       PDA Menu Line 9 =    ^^ MORE __
May 14 06:50:54 Debug:  PDA:       PDA Menu highlighted line#=8 line=''
May 14 06:50:54 Debug:  PDA:       Kicking PDA thread 19,0xb6113c00
May 14 06:50:54 Debug:  PDA:       waitForPDAMessageTypes 0x08 | 0x04, last message type was 0x0f (2 of 8)
May 14 06:50:54 Debug:  PDA:       PDA can't sleep as thread 19,0xb6113c00 is active
[... 320 lines cut ...]
May 14 06:50:57 Debug:  PDA:       PDA Menu highlighted line#=8 line=''
May 14 06:50:57 Debug:  PDA:       Kicking PDA thread 19,0xb6113c00
May 14 06:50:57 Debug:  PDA:       waitForPDAMessageTypes 0x08 | 0x04, last message type was 0x0f (2 of 8)
May 14 06:50:57 Debug:  PDA:       PDA can't sleep as thread 19,0xb6113c00 is active
May 14 06:50:57 Debug:  RS Timings:Time between packets (0.047 sec)
May 14 06:50:57 Info:   PDA:       PDA Menu Line 0 =    EQUIPMENT
[... cut ...]
May 14 06:50:57 Debug:  PDA:       Kicking PDA thread 19,0xb6113c00
May 14 06:50:57 Debug:  PDA:       waitForPDAMessageTypes 0x08 | 0x04, last message type was 0x04 (3 of 8)
May 14 06:50:57 Debug:  PDA:       waitForPDAMessageTypes: received 0x04
May 14 06:50:57 Error:  PDA:       PDA Device programmer couldn't find menu item on any page 'Pool Heater  
'
May 14 06:50:57 Error:  PDA:       PDA Device On/Off, device 'Pool Heater' not found
May 14 06:50:57 Debug:  Panl&Prog: Thread 19,0xb58fe380 (Switch PDA device on/off) finished

sfeakes, 2025-05-14:
In `AqualinkD.conf` change the button label associated with the heater `button_??_label=POOL HEAT`

microbmen, 2025-05-14:
> In `AqualinkD.conf` change the button label associated with the heater `button_??_label=POOL HEAT`

That did it.  I had a feeling it was something simple.  Thank you so much for the fast reply.

closed 2025-05-14 by microbmen