aqualinkd/AqualinkD #414: New install, Heatpump error and no temp display

mjcar505, 2025-05-16:
I just installed the system on an RS6 combo.
I know almost nothing about programming so please type slowly.
I find that i cant edit any of the config files, system says they are read only and will not let me save.
In spite of that i got most everything to work except...
The biggest issue is my heatpump (heater)  is now displaying a heater error on the rs6 keypad and it will not run, the error goes away if I shut down aqualinkd.
Also i can't see pool or spa water temp on the web interface. 
Lastly I dont need that freeze protection button in FL, any way to hide it?
Thanks for your help,
Mike C

[image]
[image]
[image]
[image]
[image]

migjaime, 2025-05-16:
https://github.com/aqualinkd/AqualinkD/wiki#Web-Config you should look over this page. It has all the details you need to update the web page.
For read-only files you need to bring up a terminal and type in something like
sudo nano /var/www/aqualinkd/config.js
and this should get you into the file to make edits. Type whatever file name you want to edit instead of /var/www/aqualinkd/config.js

sfeakes, 2025-05-16:
The above had given you the answer to how to remove "freeze protect" from the UI.  (Or you can disable freeze protect in the Jandy panel).

As for the no permission, try using `sudo` infront of what ever command you are using to edit files.  ie `sudo vi /etc/aqualinkd.com`

On the heat pump.   That's strange, we will need to see some of the logs.

 Edit config and set log_level to DEBUG.  Save config, then restart aqualinkd.
Leave that running for a minute or two, then download the log file (probably 5k lines) and post it.
Set log_level back to notice and restart aqualinkd.

mjcar505, 2025-05-17:
First thanks for taking the time to help with this.
I got the Freeze protect button removed by using   sudo nano /var/www/aqualinkd/config.js    then adding // in front of the freeze descriptor, then writing, then clearing the cache in the web viewer.
Still no luck with seeing the water temps.
I notice when i start aqualinkd the heater error comes up on the RS6 keypad and the water temp disappears from that display.

[aqualinkd.log](https://github.com/user-attachments/files/20265509/aqualinkd.log)

sfeakes, 2025-05-17:
The log you attached didn't capture AqualinkD starting up.   Can you try again, Set 5k for the download

Set log level to Debug
Save Config
Restart AqualinkD
(Wait a minute)
Download Log
[image]
Turn off Debug
Save Config

I need a log from when AqualinkD starts, so make sure the log you download has something like the following.
```
May 17 11:24:43 Notice: NetService:Received restart request!
May 17 11:24:43 Warning: AqualinkD: Stopping!
May 17 11:24:43 Warning: AqualinkD: Restarting AqualinkD!
May 17 11:24:43 Error:  RS Serial: Didn't find valid blocking serial port file descripter
May 17 11:24:43 Notice: AqualinkD: Exit!
May 17 11:24:45 Notice: AqualinkD: Starting Aqualink Daemon v2.6.5 !
May 17 11:24:45 Notice: AqualinkD: socket port                         = 80
May 17 11:24:45 Notice: AqualinkD: serial port                         = /dev/serial0
May 17 11:24:45 Notice: AqualinkD: log level                           = NOTICE
May 17 11:24:45 Notice: AqualinkD: web directory                       = /var/www/aqualinkd/
May 17 11:24:45 Notice: AqualinkD: panel type                          = RS-8 Combo (Pool & Spa)
May 17 11:24:45 Notice: AqualinkD: device id                           = 0x08
May 17 11:24:45 Notice: AqualinkD: rssa device id                      = 0x48
May 17 11:24:45 Notice: AqualinkD: extended device id programming      = YES
May 17 11:24:45 Notice: AqualinkD: extended device id                  = 0x31
May 17 11:24:45 Notice: AqualinkD: enable iaqualink                    = YES

```

mjcar505, 2025-05-17:
Here you go, hope its enough this time.

[aqualinkd.log](https://github.com/user-attachments/files/20266430/aqualinkd.log)

sfeakes, 2025-05-18:
For some reason your specific panel Rev is replying on a protocol it doesn't support. So AqualinkD is trying to connect on that protocol, and that's giving you the errors. 

Panel Rev I seems to be the problem.

To fix, you need to change the `extended_device_id` from a 0x4? to 0x3?.  To find out the last digit, to the following.

in aqmanager, run `Serial Logger` this will take a minute or two to complete.
Make note of any ID's starting with 0x40, 0x41, 0x42, 0x43 that AqualinkD states are NOT in use.

[image]

In the config, change extended_device_id to that 0x4? id

[image]

Hit `Save Config` and `Restart AqualinkD`

Can you also post the output of running Serial Logger, as that will help fix this for good.

mjcar505, 2025-05-19:
Thanks Shaun, That looks like it fixed the problem, i can see pool temps and the heater error is gone. 
Now I feel that I can do a permanent install of the hardware, probably outside near the control box,  wifi allowing...
Here is the log, funny the windows viewer shows extended device id selected as 0x40, I cleared the cache but the log says its 0x30

[image]

ts'
Notice:Serial Log:-------------------------
Notice:Serial Log:enable_iaqualink = yes
Notice:Serial Log:extended_device_id = 0x30
Notice:Serial Log:rssa_device_id = 0x48
Notice:Serial Log:device_id = 0x0a
Notice:Serial Log:-------------------------
Notice:Serial Log:Suggested aqualinkd.conf values
Notice:Serial Log:
Notice:Serial Log:ID 0x43 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:Serial Log:ID 0x42 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:Serial Log:ID 0x41 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:Serial Log:ID 0x0b is not used  <-- can use for Aqualinkd
Notice:Serial Log:ID 0x09 is not used  <-- can use for Aqualinkd
Notice:Serial Log:ID 0x49 is not used  <-- can use for Aqualinkd (RSSA ID)
Notice:Serial Log:ID 0x33 is not used  <-- can use for Aqualinkd (Prefered Extended Device ID)
Notice:Serial Log:ID 0x32 is not used  <-- can use for Aqualinkd (Prefered Extended Device ID)
Notice:Serial Log:ID 0x31 is not used  <-- can use for Aqualinkd (Prefered Extended Device ID)
Notice:Serial Log:ID 0x30 is not used  <-- can use for Aqualinkd (Prefered Extended Device ID)
Notice:Serial Log:ID 0x0a is not used  <-- can use for Aqualinkd
Notice:Serial Log:ID 0x08 is in use    <-- RS Keypad
Notice:Serial Log:ID 0x48 is not used  <-- can use for Aqualinkd (RSSA ID)
Notice:Serial Log:ID 0x40 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:Serial Log:Jandy ID's found
Notice:Serial Log:Jandy Control Panel Version :
Notice:Serial Log:Jandy Control Panel Model   :
Notice:Serial Log:RS485 interface received 800 packets in 38 seconds ( 21.05 Msg/Sec)
Notice:Serial Log:Read 800 of 800 packets
Notice:Serial Log:Read 700 of 800 packets
Notice:Serial Log:Read 600 of 800 packets
Notice:Serial Log:Read 500 of 800 packets
Notice:Serial Log:Read 400 of 800 packets
Notice:Serial Log:Read 300 of 800 packets
Notice:Serial Log:Read 200 of 800 packets
Notice:Serial Log:Read 100 of 800 packets
Notice:Serial Log:Running serial logger with 800 pakets, loglevel Notice:
Warning:AqualinkD: Starting serial_logger, this will take some time!

sfeakes, 2025-05-24:
The output you posted is from serial-logger and simply “suggested” values, not the actual values AqualinkD is using.  The suggested values are wrong for your panel rev.

closed 2025-05-24 by sfeakes