aqualinkd/AqualinkD #415: Error on updating through Upgrade AqualinkD button

migjaime, 2025-05-16:
Hi,
I am on version 2.6.3
I thought I would try the Upgrade AqualinkD button from aqmanager and I get the following error immediately.
***** AqualinkD upgrade in progress *****
Notice:NetService:Received upgrade request!
Error: AqualinkD: AqualinkD upgrade failed!
Error: AqualinkD: Upgrade error, bash exited with status: 1
Error: Upgrade:   Can not upgrade, Please fix error(s)!
Error: Upgrade1:   Can not upgrade, Please fix error(s)!
Error: Upgrade:   System Architecture not supported!
Error: Upgrade1:   System Architecture not supported!
Error: Upgrade:   System arch is amd64, this is not supported by AqualinkD
Error: Upgrade1:   System arch is amd64, this is not supported by AqualinkD

I also tried curl -fsSL https://install.aqualinkd.com | sudo bash -s -- latest
and I get the same error. Everything works right now, so nothing is broken, I was just trying to update for the hell of it. No issues with previous updates. Running Linux Mint on a Dell 5070 thin client if it matters.

sfeakes, 2025-05-16:
`Error: Upgrade: System arch is amd64, this is not supported by AqualinkD`

AqualinkD has only ever provided binaries for armhf and arm64.   So you must have compiled AqualinkD for amd64 yourself (or are using a docker).  If using docker, then that's a bug, as you shouldn't be able to see the upgrade button, since you upgrade the AqualinkD container and not AqualinkD itself.  If you compiled yourself, then simply compile again and install.

migjaime, 2025-05-16:
Yea, I compiled it myself using the steps you provided. I know I updated before but I just remembered that I switched devices and just installed it again and that is how I ended up on a newer version. My bad. This was my first time using Linux and compiling things and all the related stuff that goes with it. Lots to learn. Thanks for the quick reply and for this awesome app.

closed 2025-05-16 by migjaime