aqualinkd/AqualinkD #416: Upgrade issue with mqtt

johnwillyn, 2025-05-19:
I have been struggling with this for a day or so and thought I would ask for help.

I have been on 2.3.7 on an rpi 4 for about a year now.  I was one of the users who experienced the issue with Pentair VSP and the VSG bounce, but it has been flawless.

I want to upgrade to the latest 64 bit rPI OS (Bookworm) and installed Aqualinkd as per usual, but I am having an issue making the mqtt connection.

I have a somewhat unique configuration for mqtt; I wrote a Node based "bridge" that listens to AqualinkD over mqtt, and then sends the messages to another rPI running on my network (that also listens to a couple of specialized pool level and filling devices).

All of this was working on the older version of AqualinkD and the 32 bit OS, but the upgrade is failing to keep the connection over mqtt.  I even copied my old aqualinkd.conf file just to make sure I got everything the same, but it still fails to connect.

As others have experienced, I see it try to connect and then immediately disconnect:

Warning:NetService:MQTT Connection closed
Notice:NetService:Starting MQTT client to localhost:8883
Warning:NetService:MQTT Connection closed

Yes, I am currently configured for "localhost" which is how it worked previously, but I have also tried the IP address config, and I get the same result.

Any ideas on what might be wrong would be appreciated.  I will continue to explore this issue and hope to get it resolved.

(BTW, I hope to move to this OS so I can take advantage of PI Connect for remote management of my systems.  I don't think it is worth paying the VNC subscription fee for this capability at this point.)

Thanks in advance,
JohnL

jkosharek, 2025-05-19:
Is your MQTT broker running on the base OS or in a container and is it configured to accept web socket connection and ssl connections maybe try aqualinkd to use the SSL port that is configured? If on the base OS, does ping localhost resolve to 127.0.0.1?

johnwillyn, 2025-05-19:
The NodeJs app is running in the base OS.  I have moved away from using docker for these things to avoid some of the issues you suggest above.

ping just shows localhost, but as I mentioned, even with the explicit IP address, it does not want to start mqtt client.

johnwillyn, 2025-05-19:
I am trying a redirect (directly to the "hub" for all of my devices) now.  

It seems to be connecting to that mqtt broker, even though it is on the same OS and same NodeJS.  I will confirm all of this and update as I discover more, but I seem have a workaround in hand for now.

Thanks,
JohnL

sfeakes, 2025-05-20:
`mosquito_pub` and `mosquito_sub` are good tools to run on the pi to debug mosquito and check connections.
It kind-a sounds to me like you have a MQTT broker running on your old 32bit Pi, but not on the new 64bit Pi.
ie using localhost for the MQTT connection in AqualinkD will ONLY work if you have a MQTT broker running locally.    Again the mosquitto pub/sub tools will help you navigate that.

johnwillyn, 2025-05-20:
Thanks.  I will check out the mosquitto tools.

I have a workaround in hand by setting the listener to another PI on my network, but now I seem to get a steady set of messages on aqualinkd/SWG/Percent with value 0, even when the filter pump is off.  

I am also getting aqualinkd/SWG/fullstatus with value 255 repeated every so often.

Is this a change in behavior from previous versions?  I used to use this to flag the SWG "bounce" with the Pentair VSP, but now it comes through when not expected.

Thanks again for all of the help and support with this upgrade.

JohnL

sfeakes, 2025-05-20:
Those are changes change since 2.3.7.    full status 255 = off.  Has been there a long time though, I can;t remember the exact version / date change, but it will be in release the notes on the main page.

There were a lot os SWG changes due to the bounce issue that some had, now that's been solved, the SWG status become more stable.