aqualinkd/AqualinkD #421: Aqualink RS - Gulfstream HE 125 TA Cool Mode

mcandela0314, 2025-05-26:
I have Aqualink RS using Aqualink 3.0 for remote function. My HP is Gulfstream HE 125 TA heater/chill. My installer told me that the cool mode was only function manually and not through the Aqualink app.. Everything I've read says that the cool mode is functional through the app. If I leave the Remote T-Stat enabled, and turn the HP to "cool" the Heat Pump shows green versus red when it's on Heat mode. However, when I adjust the pool temp, the HP does not turn on. Currently, the only way to get the "cool" to work is to disable the Remote T-Stat and set the temp on the HP manually.

I read in the Aqualink RS install manual, that the heater/chill should be connected to the Relay Sockets for "Elect. Heater" and currently it is connected to the green terminal bar spots 1 and 2.

Should the "cool" function on the Gulfstream HE 125 TA be functional through the Aqualink app?

Thank you!!

[image]
[image]
[image]
[image]

sfeakes, 2025-05-26:
I think you are in the wrong place.  This is for AqualinkD, not iAqualink.
But to answer your question, you would need a RS485 connection to the heat pump for the control panel to be able to set the temperature (both heat and cool) on the Chiller.  Looks like your chiller doesn't support that type of connection.
You may get better information asking your question on https://www.troublefreepool.com/forums/

mcandela0314, 2025-05-26:
Thank you! I’ve posted in the other group.