aqualinkd/AqualinkD #424: Issues Getting Initially Installed

jmob0, 2025-06-03:
I wanted to post how I worked through some initial hiccups when getting aqualink setup. This may help others. I'll close this issue as resolved after posting.

Equipment:

* CanaKit Raspberry Pi Zero 2 W Starter MAX Kit     https://www.amazon.com/dp/B0CT1NWTHY?ref=ppx_yo2ov_dt_b_fed_asin_title
* USB-RS485-WE-1800-BT Cable, USB to RS485 Serial   https://www.amazon.com/dp/B01HI9CMT2?ref=ppx_yo2ov_dt_b_fed_asin_title
* PlusRoc Waterproof 12V/24V to 5V Converter DC-DC Step Down Module Power Adapter  https://www.amazon.com/dp/B09DGFR24W?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1

Wiring:

Don't overthink it, it really is as simple as it seems.  Data +/- from the USB connector to the 4 pin red connector on the jandy board is all that matters. Ground wasn't needed.  If the wire colors currently connected to the 4 pin connector don't match, then don't sweat it. It's the 2 middle pins no matter what color the wires are. If things don't work, swap data +/- wires....do it w/o trying to think or debug.

Stability:

Once wiring was complete, stability was the hardest thing to get squared away. PI would frequently loose connection, stop responding, or hang.  Got things stable via a watchdog, a network checking script, underclocking the CPU/RAM, & disabling wifi powersaving.  I had a 64-bit PI install which has some known issues w/ the broadcom NIC driver. If I had to do it over, I would probably use a 32-bit PI OS install instead.

Network check script - https://github.com/ltpitt/bash-network-repair-automation

Watchdog guide - https://uplab.pro/2024/12/compute-blade-auto-reboot-in-case-of-problems/

Hope this helps someone & many thanks to the author for the cobbling this together.

jmob0, 2025-06-03:
Just meant as a guide findable via search/AI.

sfeakes, 2025-06-04:
Interesting you had so many issues with stability on the Pi.  That makes me think you have a bigger problem.
Two things come to mind.
Did you flash Raspberry PI OS (with UI), or Raspberry Pi Lite (without UI)?    Lite is what you want.
Have you looked at any IO issues with the CF card you got in the kit? that might be a bottleneck causing some issues.

It may also simply be the latest OS Raspberry have put out, on a different Raspberry board but latest OS I noticed Bluetooth reaping havoc on IO, killing all those services fixed the issue.

jmob0, 2025-06-04:
Bluetooth is a possible culprit as well. It was disabled that late in the debug cycle, along w/ several other things at the same time so hard to be sure.  

No bad blocks or FS corruption on the card. It came pre flashed w/ NOOBS (this is my 1st PI experience).  After failure, the logs indicated the device was still up and functioning, but without network connectivity. Logs also had errors from the brcm driver. Wish I had saved those, but they were inline with other complaints of 64 bit brcm firmware.

I should mention this is an outdoor install, w/ direct exposure to afternoon sun, in a super hot (100+ degree F) climate.  Out of the box, CPU temps would approach 70 C.  After changes, temps are typically sub 60 w/ a peak in the low 60s.  It went from failing every few hours to auto rebooting itself once a week or so.

sfeakes, 2025-06-04:
NOOBS is your problem, it’s also no longer valid.  Simply flash Raspberry OS lite using Raspberry Imager to a CF card (maybe to a new CF card if you want to keep your setup) and call it good.