aqualinkd/AqualinkD #427: Issues with Heat Pump & AqualinkD

mal7887, 2025-06-08:
Hi Again,

We previously discussed my Jandy Heat Pump that has dual heating and chilling functions [Here](https://github.com/aqualinkd/AqualinkD/discussions/391)

I am noticing an issue. If I turn the heat pump button on, but neither Heat nor Chill is being called for (because setpoints are satisfied) The Heat Pump button on my home assistant dashboard constantly changes from and On and Off state. 

[image]

In the AqualinkD dashboard the chiller button constantly changes from on/off.

[image] 

[image]

Meanwhile in the Jandy App is shows on, but in a standby state.

[image]

Is seems like there might be a bug.

RajUmadas, 2025-06-12:
Also following along this thread. I am in the process of fixing up my aqualinkd/HA flow with my heatpump only system. So far i think things are working well.

sfeakes, 2025-06-13:
@RajUmadas   you should only need 0x70 in the mask.   That should print any messages sent to 0x70 and received from 0x70 (ie from 0x70 to 0x00).   If you use 0x00 in the mask you will get a ton of messages the panel sends out.

you might need to change the actual water and air temp to pull all status.  You can simply remove the air & water sensor wires from the panel and use resistors to change values.

RajUmadas, 2025-06-13:
sweet thanks. i was also gonna use a heat gun to mess with the temp sensors when/if i get to that. Or just record during the day and save things off.

RajUmadas, 2025-06-13:
Also just curious.. when i see a message like "0x10|0x02|0x00|0x0d|0x40|0x00|0x00|0x5f|0x10|0x03|" How does aqualinkd know this is coming FROM the heatpump. 

based on my crude understanding byte 2 is the addr.. which is set to 0x00. Where is the "source" part of this datagram that shows a "source" of "0x70"?

sfeakes, 2025-06-13:
It uses the last message.  ie `0x10|0x02|0x70` is To 0x70.  Then IF the next message is `0x10|0x02|0x00` it means it's the reply from 0x70 to 0x00. If the next message is `0x10|0x02|0x??` (something other than `0x00`) it means their was no reply.

That's the way the Jandy protocol works.

RajUmadas, 2025-06-13:
sweet... im learning so much. thanks!

mal7887, 2025-06-13:
> [@mal7887](https://github.com/mal7887) As a quick get around for the state switching between on/enabled if you set the below in aqualinkd.conf than that should fix the issue.
> 
> ```
> read_RS485_HeatPump=NO
> ```
> 
> It's the messages directly read from the heat pump that are causing the issue, reading the messages ONLY from the panel should stop the constant switching. AqualinkD will just be slower to update the actual status of the heat pump. It will be on the same time as any other Jandy interface, when usually AqualinkD is a lot quicker than Jandy interfaces since it reads messages directly.

Got it will do. Thanks!

Seems like Raj is lightyears ahead of what information I can provide, but if there is anything else I can do to help, please let me know.

mal7887, 2025-06-13:
So I made the changes you indicated above and when I restarted AqualinkD, it was stuck on "connecting to control Panel" - and it never seems to connect. I changed the configuration back, and it still gets stuck on Connecting to control panel.

It shows the SWG status, and Salt level, but nothing else.

I have restarted everything including the RS4 Panel.

Not sure what happened or caused it to break. Nothing else changed. I can control everything though the Jandy app like normal.

Pulled the debug logs

[Log.log](https://github.com/user-attachments/files/20732758/Log.log)

[aqualinkd.log](https://github.com/user-attachments/files/20732759/aqualinkd.log)

mal7887, 2025-06-13:
@sfeakes 

Oddly even though it's 'connecting' still it shows the One Touch outlet on (which it is)

[image]

sfeakes, 2025-06-13:
Looking at your logs, it's connected to the panel.   There is probably something happened that stopped that message from clearing.  If you hit re-fresh does it do away?

mal7887, 2025-06-13:
> Looking at your logs, it's connected to the panel. There is probably something happened that stopped that message from clearing. If you hit re-fresh does it do away?
> 
> EDIT. After looking again, it's not connected on the main protocol. Leave it a few minutes and it should be good.

It has been almost an hour and it still shows connecting. It's also now showing my 'Spa Jets' switch as being on, even though they are not.

RajUmadas, 2025-06-14:
Muhahaha.. I need to take a break and do some verfication.. but so far the HP commands are looking like this:

0x0c|0x00|0x00|0x00|0x00|	CMD HP Disabled
0x0c|0x0a|0x00|0x00|0x00|	CMD HP Start Heating
0x0c|0x01|0x00|0x00|0x00|	CMD HP Standby Pool
0x0c|0x02|0x00|0x00|0x00|	CMD HP Standby Spa
0x0d|0x40|0x00|0x00|	STA HP No Action
0x0d|0x48|0x00|0x00|	STA HP Heating

RajUmadas, 2025-06-14:
Ill do cooling down tomorrow, but i bet it should be what you already have in code. Based on what im seeing, processPacketToHeatPump needs to get updated with some different values. 

Im gonna do a bit more testing and really feel good about these values and maybe create a PR in a few days.

RajUmadas, 2025-06-14:
Im also just dropping raw logs and parsing things in here if you are curious. https://docs.google.com/spreadsheets/d/10dNsP-px3dEk30a_I_o7QFlijHTBwexErPaSZ-II_KA/edit?usp=sharing

sfeakes, 2025-06-14:
@RajUmadas   This is really helpful, thanks.   I'm updating the code now and will post update in next release.

I notice there is no cool in your information.  Is your heat pump a heat only and no chiller mode?

RajUmadas, 2025-06-14:
Thanks! There is no cool because i didnt try hard to capture that event. Im going to capture those tonight to make sure ive got the full set. 

But based on my reading of the code, I think you have the correct and full cool command and status.

RajUmadas, 2025-06-14:
Also, i realized my sheet might be a bit confusing, the commands tabled:
- CMD HP Pool Heat Off
- CMD HP Spa Heat Off

Has the effect of saying "Hey heat pump, you are enabled, and waiting to be called, but dont do anything yet". And it seems there are 2 different commands when the system is in pool mode of spa mode. 

I can review any PR you have as well. Im gonna try to get to some code tonight after the kids sleep.

sfeakes, 2025-06-14:
> Also,
> 
> An observation/Question. The way to turn on the Heat Pump is not intuitive in the interface. In the Jandy app, I turn on "Heat Pump" and that enables Heating and Cooling (for Pool mode, Heat Only when in Spa Mode). Both Pool Heater and Spa Heater switches are left off at all times.
> 
> Is there a way to consolidate the Heat pump into One switch and remove the unused Pool Heater and Spa Heater if the RS-485 heat Pump is enabled?
> 
> Right Now If I click Spa Heat, It turns the "Spa Heater" switch On, but if I don't turn on the Heat Pump, it will not heat. Meanwhile, I do have to click the unused switch to change the Heat Set point. The same is true for the pool.
> 
> Thoughts?

AqualinkD does exactly the same thing as the panel (it physically can't do anything different).  So you should just turn the heat pump/chiller on in AqualinkD (or HA) and it'll do the same thing as the Jandy app.

RajUmadas, 2025-06-15:
> [@RajUmadas](https://github.com/RajUmadas) I've just posted 2.6.8 (Dev), take a look and see if that's correct from your findings. You can also download the binary and test it.

Noob question, whats the best way to try and test out this version. I dont see anything in tagged releases. 

Also, i probably wont get to this until monday, so no rush.

RajUmadas, 2025-06-15:
Im also guessing since you didnt tag a dev release, curl -fsSL `https://raw.githubusercontent.com/aqualinkd/AqualinkD/master/release/remote_install.sh | sudo bash -s -- development` wont work to test out the dev build. correct?

RajUmadas, 2025-06-15:
Sorry for being so chatty on thread. Looks like the install script did work based on the UI. The script output confused me.

RajUmadas, 2025-06-15:
This is also a quick screenshot of my HA dashboard im using for the pool.... just to get an idea. 
[image]

mal7887, 2025-06-15:
> > Also,
> > An observation/Question. The way to turn on the Heat Pump is not intuitive in the interface. In the Jandy app, I turn on "Heat Pump" and that enables Heating and Cooling (for Pool mode, Heat Only when in Spa Mode). Both Pool Heater and Spa Heater switches are left off at all times.
> > Is there a way to consolidate the Heat pump into One switch and remove the unused Pool Heater and Spa Heater if the RS-485 heat Pump is enabled?
> > Right Now If I click Spa Heat, It turns the "Spa Heater" switch On, but if I don't turn on the Heat Pump, it will not heat. Meanwhile, I do have to click the unused switch to change the Heat Set point. The same is true for the pool.
> > Thoughts?
> 
> [@mal7887](https://github.com/mal7887) AqualinkD does exactly the same thing as the panel (it physically can't do anything different). So you should just turn the heat pump/chiller on in AqualinkD (or HA) and it'll do the same thing as the Jandy app.

@sfeakes I guess what I was hoping for was a combined thermostat for the pool, if the heat pump is configured it has both heat and cool set points in one place.

Similar to how my nest thermostat looks below.

[image]

RajUmadas, 2025-06-17:
Ah i get it. I need to use the spa and pool heat elements to see the change... Looks like it working @sfeakes 

[image]

sfeakes, 2025-06-17:
@mal7887. You can do that in HA yourself.    I don’t think I will be putting that into the AqualinkD web UI anytime soon, well not until I completely understand all the different Heatpump options and configurations.  But in HA you can make your own control to do that, all the information on MQTT messages is documented. 

> 
> [@sfeakes](https://github.com/sfeakes) I guess what I was hoping for was a combined thermostat for the pool, if the heat pump is configured it has both heat and cool set points in one place.
> 
> Similar to how my nest thermostat looks below.
> 
> [image]

sfeakes, 2025-06-29:
@RajUmadas I just posed v2.6.8, that has the fix.