aqualinkd/AqualinkD #438: Incorrect Temperature Handling When PDA Only Panel Is Set to Celsius

mcharman77, 2025-06-23:
First, I’d like to thank you for such an excellent project and all the time and effort that’s gone into AquaLinkD. It’s been running reliably on my PDA-only panel setup, and I really appreciate the support for older Jandy systems.

I’ve run into a minor but persistent issue related to temperature units. AquaLinkD appears to assume that the panel is using Fahrenheit, but in my case, I’d prefer to use Celsius on the PDA panel. Unfortunately, when I set the PDA to Celsius, AquaLinkD continues to treat the reported values as if they are in Fahrenheit — resulting in incorrect temperature readings in Home Assistant. For example, Home Assistant double-converts the value, thinking it's in Fahrenheit even when it's already Celsius, leading to inaccurate results.

Additionally, when the PDA is set to Celsius, the minimum heater setpoint is limited to 37°C, which is unnecessarily high. This further complicates practical use.

**Feature Request:**
Would it be possible to add a configuration option (e.g., temp_unit: C) in aqualinkd.conf that explicitly tells AquaLinkD whether the PDA/panel is operating in Celsius or Fahrenheit? This way, the software could properly handle the values and avoid misinterpreting the units — especially in setups where the physical PDA is set to Celsius.

This would make AquaLinkD much more usable for international users and help maintain accuracy across MQTT integrations and Home Assistant.
Thanks again for your hard work on this great project!

mcharman77, 2025-06-24:
Thank you so much for looking into this and considering it for a future release.

To be honest, I find the panel models and revisions a bit confusing. I don’t have iAqualink, but I’ve attached some photos of my PDA. Based on what I can see, I appear to have the following:

PDA-P4 Only

PDA version 7.0

J-Box version 4.0

Handheld version 3.0

I’m able to change the temperature unit by navigating through the following menu path:
MENU → SYSTEM SETUP → DEGREES C/F

I hope this information is helpful.

Thanks again!

[image]
[image]
[image]
[image]
[image]

sfeakes, 2025-06-29:
@mcharman77   I just posted version 2.6.8, that should have the fix for you.

mcharman77, 2025-06-30:
Just gave it a try. Both the web interface and Home Assistant appear to be working as expected:) Thank you once again!