aqualinkd/AqualinkD #440: Light devices and virtual buttons possible?

robbydyer, 2025-06-28:
Is it possible to do light control (telling the Aqualink controller which color to use, not controlling with AqualinkD) with virtual buttons mapped to a device? I have two banks of lights defined as iAQ touch devices ("POOL LIGHTS" and "BUBBLERS" in the log snippet).

I'm able to successfully turn the lights on/off, but I can't figure out how (or if) I can switch the light colors.

My config for the virtual buttons. I'm not sure the `_lightMode` settings for these are valid. 
```
virtual_button_02_label=POOL LIGHTS
virtual_button_02_lightMode=2

virtual_button_03_label=BUBBLERS
virtual_button_03_lightMode=2
```

```
Jun 27 22:35:35 Debug:  iAQ Touch: Turning IAQ SEND on
Jun 27 22:35:35 Info:   iAQ Touch: Page: Devices (rev Yg) | 0x0a
Jun 27 22:35:35 Debug:  iAQ Touch: Button 00|      Filter Pump ON   | type=0x01 | state=0x01 | unknown=0x00 | keycode=0x11
Jun 27 22:35:35 Debug:  iAQ Touch: Button 01|      Filter Pump ADJ  | type=0x01 | state=0x01 | unknown=0x00 | keycode=0x12
Jun 27 22:35:35 Debug:  iAQ Touch: Button 02|        Pump 4100 ADJ  | type=0x01 | state=0x00 | unknown=0x00 | keycode=0x13
Jun 27 22:35:35 Debug:  iAQ Touch: Button 03|            Temp1 ENA  | type=0x01 | state=0x03 | unknown=0x00 | keycode=0x14
Jun 27 22:35:35 Debug:  iAQ Touch: Button 04|            Temp2 OFF  | type=0x01 | state=0x00 | unknown=0x00 | keycode=0x15
Jun 27 22:35:35 Debug:  iAQ Touch: Button 05|        Heat Pump OFF  | type=0x01 | state=0x00 | unknown=0x00 | keycode=0x16
Jun 27 22:35:35 Debug:  iAQ Touch: Button 06|          Feature OFF  | type=0x01 | state=0x00 | unknown=0x00 | keycode=0x17
Jun 27 22:35:35 Debug:  iAQ Touch: Button 07|             Aux2 OFF  | type=0x01 | state=0x00 | unknown=0x00 | keycode=0x18
Jun 27 22:35:35 Debug:  iAQ Touch: Button 08|             Aux3 OFF  | type=0x01 | state=0x00 | unknown=0x00 | keycode=0x19
Jun 27 22:35:35 Debug:  iAQ Touch: Button 09|        Extra Aux OFF  | type=0x01 | state=0x00 | unknown=0x00 | keycode=0x1a
Jun 27 22:35:35 Debug:  iAQ Touch: Button 10|      POOL LIGHTS OFF  | type=0x01 | state=0x00 | unknown=0x00 | keycode=0x1b
Jun 27 22:35:35 Debug:  iAQ Touch: Button 11|         BUBBLERS OFF  | type=0x01 | state=0x00 | unknown=0x00 | keycode=0x1c
Jun 27 22:35:35 Debug:  iAQ Touch: Button 12|          All OFF OFF  | type=0x01 | state=0x00 | unknown=0x00 | keycode=0x1d
Jun 27 22:35:35 Debug:  iAQ Touch: Button 13|                       | type=0x00 | state=0x00 | unknown=0x00 | keycode=0x1e
Jun 27 22:35:35 Debug:  iAQ Touch: Button 14|                       | type=0x00 | state=0xff | unknown=0x00 | keycode=0x1f
Jun 27 22:35:35 Debug:  iAQ Touch: Button 15|                       | type=0x00 | state=0xff | unknown=0x00 | keycode=0x00
Jun 27 22:35:35 Debug:  iAQ Touch: Button 16|                       | type=0x00 | state=0xff | unknown=0x00 | keycode=0x00
```

And confirmed in startup my lights are mapped to Aux_V2 and Aux_V3:
```
Jun 27 22:29:37 Notice: AqualinkD: Button Filter_Pump   = label Filter Pump     | VSP ID 0xe0 | PumpID 1 | Filter Pump
Jun 27 22:29:37 Notice: AqualinkD: Button Aux_1         = label Slide pump      | VSP ID 0xe1 | PumpID 2 | Pump 4100
Jun 27 22:29:37 Notice: AqualinkD: Button Aux_2         = label Aux 2           |
Jun 27 22:29:37 Notice: AqualinkD: Button Aux_3         = label Aux 3           |
Jun 27 22:29:37 Notice: AqualinkD: Button Pool_Heater   = label Temp1 Heater    |
Jun 27 22:29:37 Notice: AqualinkD: Button Spa_Heater    = label Temp2 Heater    |
Jun 27 22:29:37 Notice: AqualinkD: Button Extra_Aux     = label Extra Aux       |
Jun 27 22:29:37 Notice: AqualinkD: Button Aux_V1        = label Heat Pump       | Chiller     |
Jun 27 22:29:37 Notice: AqualinkD: Button Aux_V2        = label POOL LIGHTS     |
Jun 27 22:29:37 Notice: AqualinkD: Button Aux_V3        = label BUBBLERS        |
```

I tried sending MQTT messages to `aqualinkd/Aux_V2/color/set` just to see, but it results in an error:
```
Error: Panl&Prog: Light mode control not configured for button 8
```

sfeakes, 2025-06-28:
Why are you trying to use virtual button for light mode?
The Jandy panel can only support light mode on a specific circuit, and AqualinkD is the same.  Simply set the light mode against the actual circuit (button) that has the lights attached to it, then send the MQTT message to that button / circuit.   Or long priest in the web UI on that button will bring up the light mode options.

This is assuming you have an RS panel and not PDA.

robbydyer, 2025-06-29:
I could be mistaken, but I don't know if the lights are mapped to a button. When I turn them on in the iAqualink app from the "Equipment" tab, none of the lights inisde my panel come on. Pushing all the Aux buttons on the panel doesn't turn them on either. Perhaps I'm missing something in Aqualink that maps the two light devices to a button? I believe it's an RS8 panel (I see connectors up to aux 7), but only 1-3 have anything connected. Aux 1 is an additional pump for a water feature.

[image]

robbydyer, 2025-06-29:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

sfeakes, 2025-06-29:
I was wondering if Jandy's "infinite watercolors" would be RS485 controlled. Finally someone came out with RS485 controlled LED's.
I'm going to need a lot of information from you to get this working.   First off, I'd like to know the ID's your system is using.  Can you turn on your lights. Then run the `Serial Logger` in `aqmanager`, wait for that to finish and spit out the ID's, download the log and post it please.

[image]

robbydyer, 2025-06-29:
[aqualinkd.log](https://github.com/user-attachments/files/20970166/aqualinkd.log)

sfeakes, 2025-07-06:
Thanks so much for that.   I'd like to get some detailed RS485 information now if you wouldn't mind.

Would you mind doing the following.

- Make sure lights are off
- In Aqmanager can you set the RS Serial filter to 0x33,0xA3,0xF0 and then set it to  on. (must set the filter first)
- Using the iAqualink web interface (not mobile app), can you turn the light on to a specific color mode.
- wait a few seconds
- turn the light to a different mode
- wait a few seconds
- turn the brightness to 50%
- wait a few seconds
- turn the brightness to 25%
- wait a few seconds
- turn light off
- download the log (10k lines) and post it.  (also letting me know the color modes)

aqmanager should look like below.
[image]

sfeakes, 2025-07-07:
Oh, and one other thing (in a different log).   Can you set the same RS filters.  Then turn the light to fully RED, then GREEN, then BLUE.

Thanks.

robbydyer, 2025-07-08:
Color change and then switched off (Alpine white to start -> change to "Sky Blue" -> change to "Cobalt Blue") : [color-change-off.log](https://github.com/user-attachments/files/21112215/color-change-off.log)

For RGB, I don't have a way to set pure RGB (i.e. 255-0-0, 0-255-0, etc), the UI gives a custom color wheel. I did my best to select red, green, blue in that order: [rgb.log](https://github.com/user-attachments/files/21112191/rgb.log)

Also, I have not found a way to control the lights at all (other than on/off) via the web interface, only the mobile app. Clicking the "Pool lights" button in the Devices list just toggles them on/off.

sfeakes, 2025-07-09:
Thanks for posting that, very helpful.

> Also, I have not found a way to control the lights at all (other than on/off) via the web interface, only the mobile app. Clicking the "Pool lights" button in the Devices list just toggles them on/off.

This confirms what I was seeing in the logs, as to why I thought we were missing something.  If this is the case, then we may be out of luck trying to control the lights. Jandy have limited the iAqualink 2/3 protocol to support only 1 device, and if iAqualink is the only protocol that can change the light color on RS485 lights, then there is no way AqualinkD can do it, unless you remove their WiFi adapter so AqualinkD can use the iAqualink protocol.
There may be a way round it, I'll post some updated code in a week or so and see if that works.