aqualinkd/AqualinkD #442: PDA Freezes When Using AqualinkD and Physical PDA Simultaneously (v2.6.8)

mcharman77, 2025-07-01:
I’m wondering if it’s possible to use AqualinkD and the physical PDA at the same time?

I'm currently running AqualinkD version 2.6.8. According to the release notes for version 2.5.0 (Nov 16, 2024), this functionality should be supported. However, in my setup, it doesn’t seem to work as expected. When I turn the physical PDA on, it initially connects and AqualinkD appears to continue working, but shortly after, the PDA freezes.

From the 2.5.0 release notes, I saw the following update:
•	PDA panel Rev 6.0 or newer that do not have a Jandy iAqualink device attached can use the AqualinkTouch protocol rather than PDA protocol.

My panel details when the PDA is powered on are:
•	PDA-P4 Only
•	PDA version 7.0
•	J-Box version 4.0
•	Handheld version 3.0

I believe this suggests I have PDA panel Rev 7.0, so it should support the newer protocol, but not totally sure.
Also from the release notes:
•	Use device_id=0x33 in aqualinkd.conf.

However, I don’t see an option to select device_id=0x33 from the dropdown in the configuration editor — it’s not listed.

I've also attached my config file for reference.

[aqualinkd.txt](https://github.com/user-attachments/files/20989774/aqualinkd.txt)

Thanks in advance for your help.

sfeakes, 2025-07-01:
You must have something off in your config.  Looks like you may have multiple entries os something.  Check there are no duplicates, and that the below are set.

```
panel_type=PDA-8 Combo        <-  Change to 4 or 6 depending on PDA.
device_id=0x33
```

mcharman77, 2025-07-13:
In version 2.6.8, when I set panel_type=PDA-4 Only, AqualinkD automatically changes it to panel_type=PDA-8 Only. The same thing happens when I set panel_type=PDA-4 Combo — it gets changed to panel_type=PDA-8 Combo.

As a test, I reverted to v2.6.7, which does allow me to set panel_type=PDA-4 Combo (I also tried panel_type=PDA-4 Only), but AqualinkD still throws the following errors when I set device_id=0x33:

aqualinkd-1  | Error:   AqualinkD: Device ID 0xff does not match PDA panel, please check config!

aqualinkd-1  | Error:   AqualinkD: Config entry 'device_id=0x33',  0x33 is not valid
aqualinkd-1  | Error:   AqualinkD: Unknown config parameter 'device_id=0x33'
aqualinkd-1  | Error:   AqualinkD: Device ID 0xff does not match PDA panel, please check config!
aqualinkd-1 exited with code 1

I don’t believe there are duplicate entries in my config file, but I’ve attached it in case you have time to take a look.

[aqualinkd.txt](https://github.com/user-attachments/files/21201842/aqualinkd.txt)

sfeakes, 2025-07-13:
There is a bug, sorry about that.  if you use `PDA-PS4 Only` that should work for setting the panel.

Can you post the output of serial_logger, I'd like to see exactly how the panel formats that size / type.

Let me look into the 0x33 issue.

mcharman77, 2025-07-13:
Yes I can post the log, but please clarify how you would like me to config (panel type, device_id 0x33 ?). Also what AqualinkD version? Just not quite sure.

sfeakes, 2025-07-13:
Use 0x60 for the moment until I have done some more debugging.   Use latest version.

mcharman77, 2025-07-14:
Please see the attached log file. I'm not sure if I captured everything you need, so please let me know if I didn't.
I basically enabled "Debug Serial," waited for a while, did a restart, then waited another minute or so before downloading the log file.

[aqualinkd (3).log](https://github.com/user-attachments/files/21207046/aqualinkd.3.log)

mcharman77, 2025-08-08:
Hi there,

Thank you for the update in v2.6.9 to allow device_id 0x33. Unfortunately, I think it’s led to some strange behaviour—at least on my setup. I’ve attached my config and a log file for reference.

Here’s what I’ve noticed:

The Filter Pump icon on the web interface no longer displays RPM and Watts.

When I click the Sheer Descent icon, the Pool Lights turn on (at least according to the web interface icon).

When I click the Pool Light icon, the Sheer Descent turns on (at least according to the web interface icon).

I can no longer control the Heater (it doesn’t turn on). However, it appears that Spa_Heater actually turns the heater on (this can be seen on the PDA) with the pool temperature displayed in the Pool_Heater icon.

The PDA is also transitioning to unexpected places in the menu system after I press an icon on the web interface and then push a button on the PDA.

[aqualinkd.conf.txt](https://github.com/user-attachments/files/21688574/aqualinkd.conf.txt)
[aqualinkd (6).log](https://github.com/user-attachments/files/21688573/aqualinkd.6.log)

closed 2025-12-18 by github-actions[bot]