aqualinkd/AqualinkD #446: No Spa Mode button in iOS Chrome

markstubb, 2025-08-03:
AqualinkD 2.6.9
For some reason, the Spa Mode button doesn't show up in Chrome (139.0.7258.60) on iOS (18.5). Works fine on the desktop browsers, also works fine in Safari on the same device.
Desktop

[image]

Safari Mobile

[image]

Chrome Mobile

[image]

markstubb, 2025-08-03:
turns out it was some sort of weird caching issue?  I cleared the cache in the chrome mobile browser and the button reappeared.  I tried multiple refreshes, tab closures, etc, but had to actually delete the cache on the device and now it works.  Have never had to do that before, oddly enough.

closed 2025-08-03 by markstubb

sfeakes, 2025-08-04:
AqualinkD uses heavy caching to save load,  so every time you change config it’s worth clearing cache.  If you just did an update, then the last few versions have changed the web configuration, so that’s probably the issue.